sn4k3/UVtools #369: [BUG] Anycubic Mono 4K incorrect display size in PursaSlicer

nviennot, 2021-12-13:
In PursaSlicer -> Printer Settings,
Bed shape -> Set ... -> bed size is set to be 132.9 x 80mm. This is correct.

However, the display size is incorrectly set to 132.9 x 80mm. It should be 134.4 x 84mm.
The reason is that each pixel is 35um. The display resolution is 3840x2400. Multiplying the display resolution by the pixel size gives 134.4 x 84mm.

Currently, the prints are coming out stretched. Circular shapes are coming out as ovals.
Let me know how I can help fix this.

Nico

sn4k3, 2021-12-13:
Ok, i copied the settings from Photon workshop. If that's the case other printers will share same problem... I will fix it.

nviennot, 2021-12-13:
Awesome! Thank you Tiago

sn4k3, 2021-12-18:
Fixed

closed 2021-12-18 by sn4k3

nviennot, 2022-01-05:
Hello @sn4k3, another nitpick: The origin of the build plate is set to (0,0), but in the case of the Mono 4K, the build plate is centered in relationship to the screen, so the origin should be (-0.75, -2).
This is coming from `-(display_width - bed_shape_width)/2`, same for height.

sn4k3, 2022-01-05:
It should not cause any trouble, as long people dont slice whole lcd area.
Do you notice any diference between a origin of 0 and the correct? I guess it will shift model in the bed by that amount?

nviennot, 2022-01-05:
I did a print where I used the entire bed shape. It mattered (2mm off is problematic)