sn4k3/UVtools #371: libdl: cannot open shared object file: No such file or directory (Fedora 35)

mcdviii, 2021-12-14:
I'm experiencing the following failure when trying to start the application via `dotnet UVtools`, `./UVtools.sh` or `./UVtools`

```
./UVtools                                                                                                                                            1 ↵
Unhandled exception. System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'dl' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libdl: cannot open shared object file: No such file or directory
   at Emgu.Util.Toolbox.Dlopen(String dllname, Int32 mode)
   at Emgu.Util.Toolbox.LoadLibrary(String dllname)
   at Emgu.CV.CvInvoke.LoadUnmanagedModules(String loadDirectory, String[] unmanagedModules)
   at Emgu.CV.CvInvoke.DefaultLoadUnmanagedModules(String[] modules)
   at Emgu.CV.CvInvoke..cctor()
   --- End of inner exception stack trace ---
   at Emgu.CV.CvInvoke.Init()
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\App.axaml.cs:line 70
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run() in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 166
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.X11.X11PlatformThreading.CheckSignaled() in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 164
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 244
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 61
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 126
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 175
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 47
[1]    103186 IOT instruction (core dumped)  ./UVtools
```
Unsure what I need to be looking for in order to resolve this particular dependency. Any help is appreciated.

sn4k3, 2021-12-14:
You need this dependencies: https://github.com/sn4k3/UVtools#rhelfedoracentos
And download the rhel package on UVtools downloads, which one you used?

mcdviii, 2021-12-20:
I'm working from the RHEL release v2.26.0

I've followed the steps outlined in the README, but have run into the following error after initiating `./cmake_configure`

[https://pastebin.com/GLRxyVEW](url)

> The current .NET SDK does not support targeting .NET 6.0

sn4k3, 2021-12-20:
You are trying to compile the OpenCV that's it?
What commands are you running?
Your pastebin does not work

mcdviii, 2021-12-20:
Here is the log again, Pastebin is autoremoving it for some reason:

[https://www.toptal.com/developers/hastebin/itanonoluz.sql](https://www.toptal.com/developers/hastebin/itanonoluz.sql)

I've installed all necessary deps & ran the commands outlined to build the dependances under the `emgucv` directory (which I've created under UVTools, and cloned the repo to as directed in the example).

sn4k3, 2021-12-20:
your problem: `/usr/lib64/dotnet/sdk/5.0.206/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(141,5): error NETSDK1045: The current .NET SDK does not support targeting .NET 6.0.  Either target .NET 5.0 or lower, or use a version of the .NET SDK that supports .NET 6.0. [/home/smaee/Source/UVtools/emgucv/Emgu.CV.Test/Emgu.CV.Test.Netcore/Emgu.CV.Test.Netcore.csproj]`

That mean you have .NET 6.0 which is out of requirements, you need NET 5.0. But this compilations are not required, what you need is already generated: `[ 92%] Linking CXX shared library ../../../../../libs/libcvextern.so` --> That is all you need now, copy that file to UVtools directory