sn4k3/UVtools #376: Problem opening a CXDLP file (Creality Halot Sky)

bordermultimedia, 2021-12-19:
## System
 - OS: Windows 10 x64 v21H2
 - Slicer: Lychee Slicer
 - Printer: Creality Halot Sky
 - UVtools version: 2.27.0 (previous version also showed the error)

```
1. Go to "UVtools -> Help -> About"
2. Click on "Copy information"
3. Paste that infomation bellow this line:
UVtools v2.27.0
Operative system: Microsoft Windows 10.0.19044 X64
Processor: 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz
Processor cores: 16
Memory RAM: 53.24 / 127.88 GB
Runtime: win10-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.8
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary) (On this)
    WA: 3840 x 2100    UA: 2560 x 1400
2: 1680 x 1050 @ 100%
    WA: 1680 x 1010    UA: 1680 x 1010

Path: C:\Program Files\UVtools

General configuration for OpenCV 4.5.4 =====================================
  Version control:               4.5.4-235-g27567928d7

  Extra modules:
    Location (extra):            D:/bb/cv_x64/build/build_x64/../opencv_contrib/modules
    Version control (extra):     4.5.4-61-gc33e5b60

  Platform:
    Timestamp:                   2021-10-27T09:08:07Z
    Host:                        Windows 10.0.22000 AMD64
    CMake:                       3.21.3
    CMake generator:             Visual Studio 16 2019
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
    MSVC:                        1929
    Configuration:               Debug Release MinSizeRel RelWithDebInfo

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe  (ver 19.29.30136.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MD /O2 /Ob2 /DNDEBUG 
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MDd /Zi /Ob0 /Od /RTC1 
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP   /MD /O2 /Ob2 /DNDEBUG 
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP /MDd /Zi /Ob0 /Od /RTC1 
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO 
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL 
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          wsock32 comctl32 gdi32 ole32 setupapi ws2_32 D:/bb/cv_x64/build/build_x64/install/lib/freetype.lib D:/bb/cv_x64/build/build_x64/install/lib/harfbuzz.lib D:/bb/cv_x64/build/build_x64/install/lib/libhdf5.lib vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc

  OpenCV modules:
    To be built:                 alphamat aruco barcode bgsegm bioinspired calib3d ccalib core datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hdf hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking ts video videoio videostab viz wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    python_bindings_generator python_tests world
    Disabled by dependency:      -
    Unavailable:                 cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev cvv java julia matlab ovis python2 python3 sfm
    Applications:                perf_tests
[... 173 lines cut ...]

## To Reproduce
Steps to reproduce the behavior:
1. Export a CXDLP file using Lychee and Creality Halot Sky as the printer
2. Open the file with UV tools
3. The error message appears

## Expected behavior
File being opened

## Screenshots
[image]

## Files
Ive tried a couple of files, this is one of them, using no supports:
[soportes-tapas.zip](https://github.com/sn4k3/UVtools/files/7741711/soportes-tapas.zip)
This was the output:
[test.zip](https://github.com/sn4k3/UVtools/files/7741713/test.zip)

 
## Exceptions Log file
No log file was generated

## Additional context
I have no problems when generating and opening files for my Phrozen Sonic Mini

sn4k3, 2021-12-19:
Lychee slicer is outputing new file version.
Does it print if you attempt to?

sn4k3, 2021-12-22:
Fixed

closed 2021-12-22 by sn4k3