sn4k3/UVtools #378: NullReferenceException

sigxcpu76, 2021-12-22:
## System
 - OS: KDE Neon (Ubuntu 20.04/focal based)
 - Slicer: HALOT BOX
 - Printer: Creality HALOT ONE
 - UVtools version: 2.27.1

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

UVtools v2.27.1
Operative system: Linux 5.15.9-051509-generic #202112161611-Ubuntu SMP Thu Dec 16 16:23:18 UTC 2021 X64
Processor: AMD Ryzen 9 5900X 12-Core Processor
Processor cores: 24
Memory RAM: 55.99 / 62.78 GB
Runtime: neon.20.04-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.11
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 100% (Primary)
    WA: 3762 x 2160    UA: 3762 x 2160

Path: /home/sigxcpu/Downloads/UVtools_linux-x64_v2.27.1

General configuration for OpenCV 4.5.4 =====================================
  Version control:               4.5.4-235-g27567928d7

  Extra modules:
    Location (extra):            /var/lib/buildbot/workers/worker/cv_ubuntu/build/opencv_contrib/modules
    Version control (extra):     4.5.4-61-gc33e5b60

  Platform:
    Timestamp:                   2021-10-26T15:12:54Z
    Host:                        Linux 4.19.104-microsoft-standard x86_64
    CMake:                       3.16.3
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3
    Dispatched code generation:  SSE4_1 SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_1 (15 files):         + SSSE3 SSE4_1
      SSE4_2 (1 files):          + SSSE3 SSE4_1 POPCNT SSE4_2
      FP16 (0 files):            + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 AVX
      AVX (4 files):             + SSSE3 SSE4_1 POPCNT SSE4_2 AVX
      AVX2 (30 files):           + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (5 files):      + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /usr/bin/c++  (ver 9.3.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--exclude-libs,libippicv.a -Wl,--exclude-libs,libippiw.a   -Wl,--gc-sections -Wl,--as-needed  
    Linker flags (Debug):        -Wl,--exclude-libs,libippicv.a -Wl,--exclude-libs,libippiw.a   -Wl,--gc-sections -Wl,--as-needed  
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          /var/lib/buildbot/workers/worker/cv_ubuntu/build/platforms/ubuntu/20.04/build/install/lib/libhdf5.a vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy Iconv::Iconv dl m pthread rt
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc ippiw ippicv

[... 207 lines cut ...]
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 132
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 187
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 47
[1]    3669261 abort (core dumped)  ./UVtools
```

## To Reproduce
Steps to reproduce the behavior:
1. Open the model
2. Click on 'Issues'
4. See error

## Expected behavior

No crash.

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2021-12-22:
Try downgrade to v2.27.0 and reply if it solved

sigxcpu76, 2021-12-22:
It throws an error when opening the file on 2.27.0

[image]

sn4k3, 2021-12-22:
> It throws an error when opening the file on 2.27.0
> 
> [image]

But your printer is Creality HALOT ONE which use .cxdlp, why are you using a .zip file?

sigxcpu76, 2021-12-22:
You are right. I've prepared the ZIP file to attach on GitHub, but it failed to attach. By mistake, I've picked that one.

Yes, clicking on Issues show the message about "... if they are calculated". No more crash.

sn4k3, 2021-12-22:
Ok, so keep in that version for now if that work ok. I will try to redirect this problem

sigxcpu76, 2021-12-22:
Thanks for support.

i-hayes, 2022-01-07:
I'm having the same problem with or without a file loaded. Tried your suggestion to down grade to 2.27.0 and so far program appears stable and can load and analyse files.

UVtools v2.27.5
Operative system: Linux 5.3.18-59.37-default #1 SMP Mon Nov 22 12:29:04 UTC 2021 (d10168e) X64
Processor: Intel(R) Core(TM) i7 CPU         950  @ 3.07GHz
Processor cores: 8
Memory RAM: 15.78 / 17.56 GB
Runtime: opensuse-leap.15.3-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.11
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1034    UA: 1920 x 1034
2: 1600 x 900 @ 100%
    WA: 1600 x 854    UA: 1600 x 854
3: 1920 x 1080 @ 100%
    WA: 1920 x 1034    UA: 1920 x 1034

Path: /home/ihayes/bin/UVtools_linux-x64

sn4k3, 2022-01-27:
Can you guys confirm if this is fixed on new release?

yaqwsx, 2022-02-01:
I just tried 2.27.7 and I did not observe a crash.

Lenbok, 2022-02-06:
Version 2.27.7 crashes for me still, on Ubuntu 20.04.

sn4k3, 2022-02-14:
Fixed