sn4k3/UVtools #387: [BUG]

SaberOfBorg, 2021-12-29:
## System
 - OS: [e.g. Windows 10 x64 v20H2]
 - Slicer: Lychee
 - Printer: Anycubic Mono X (4k)
 - UVtools version: 2.27.4

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:
UVtools v2.27.4
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz
Processor cores: 12
Memory RAM: 6.99 / 15.81 GB
Runtime: win10-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.11
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1200 @ 100% (Primary)
    WA: 1920 x 1200    UA: 1920 x 1200
2: 1920 x 1200 @ 100% (On this)
    WA: 1920 x 1200    UA: 1920 x 1200
3: 1080 x 1920 @ 100%
    WA: 1080 x 1920    UA: 1080 x 1920

Path: D:\!System_Maintenance\!App Install Files\3D Printer\UV Tools v2.27.4

```

## Describe the bug
"Exposure Time Finder" file prints with out-of-sequence exposures on Anycubic Photon Mono X

I used the "Exposure Time Finder" to create a series of 14 test areas in the range of 1.5sec to 4.1sec (in 0.2sec increments) in order to calibrate my new printer to its resin.  According to UV Tools the test areas were to represent exposure times of:
2.7   2.5   2.3   2.1   1.9   1.7   1.5
4.1   3.9   3.7   3.5   3.3   3.1   2.9

I put a piece of copy paper on the LCD screen and let the printer pretend to 'print' the test file (see attached) so that I could observe that the printer could actually do the multiple exposures per layer.  I read the tiny text (thru the copy paper) for each test area to confirm their arrangement on the screen matched what UV Tools had shown me, and it did.  As the printer went through illuminating each of the test areas I used a stop-watch app to time the "light on" for each area, and I found (after letting it run 3 cycles to confirm my measurements) the zones were being exposed as:
2.5   2.3   2.1   1.9   1.7   1.5   4.1
3.9   3.7   3.5   3.3   3.1   2.9   2.7

All of the test zones' exposure times seem to be shifted-over by 1 test cell.  The zone that will have the "1.5s" text printed on it will actually be exposed for 4.1sec :-(  I will have to carefully peel the test pieces off the Build Plate and label them by-hand with  marker (and ignore the text printed on the pieces).

Is this a problem with my printer, or a problem with the file created by UV Tools?

## Files

[Exposure Test Anycubic Grey.zip](https://github.com/sn4k3/UVtools/files/7789780/Exposure.Test.Anycubic.Grey.zip)
>

sn4k3, 2021-12-29:
Nothing wrong with the file, both uvtools and photonworkshop agrees but the problem may be in firmware... If you can record a video with the problem visible.
Also read that: https://www.facebook.com/groups/uvtools/posts/596870478089126

SaberOfBorg, 2021-12-30:
Tiago:
Many thanks for the super-fast response to this.
Attached are several pictures for you to review; sadly the videos are too large to attach.
I've included a picture showing that my printer has the latest firmware (v3.5.4).
There's a picture showing the full-up image of the test tiles (taken during 1 of the burn-in layers).
There's also pictures zooming-in on each of the corners of the test tile pattern so you can more easily read the intended exposure times.
Videos show the initial 6 burn-in layers (the "Layers 4 and 5" video), some strange/intermediary 7th layer, and then a representation of the start of the series of "exposure test" layers ("Layer 8").  It's a shame I can't attach them, as the MP4 files are 142MB, 412MB, and 580MB.  I'll attempt to re-code them in a more size-efficient format (I'm sure they will suffer visually, but, at least you'll be able to see the lighted 'blobs' and measure their durations) and then post later today.

What I find interesting/odd about layer #7 is that the intended 1.5s tile is exposed for about 18s, then the 1.7s tile is exposed for about 16s, further tile exposures constantly decreasing in duration until it reaches the intended 4.1s tile.  What's-up with this strange sequence? Why wouldn't layer 7's exposure times be the same as all the subsequent layers?

As you will see from the "Layer 8" video is that the 1st (upper right) tile to be exposed, which should be 1.5s, is actually "on" for 4.1sec . . . then the 1.7s tile is exposed for 1.5s, the 1.9s tile is exposed for 1.7s, etc.  Everything seems to be shifted-by-one.

[image]
[image]
[image]
[image]
[image]
[image]

SaberOfBorg, 2021-12-30:
Here are the videos . . hopefully they are still useful despite my needing to crunch-down the resolution in order to make them fit in 10MB (and yet have decent time accuracy...not 'jerky').

Question: When the 8th and beyond layers are being exposed, and the various tiles are being exposed for their intended times, why does the Build Plate move up/down?  I would think that for a given layer (regardless of how many test tiles there are) the resin is already in-place and ready to be exposed.  What's the purpose of lifting/retracting the Build Plate between test tiles (only to put it back to, presumably, the same Z location)?

[Layers 4 and 5.zip](https://github.com/sn4k3/UVtools/files/7794069/Layers.4.and.5.zip)
[Layer 7.zip](https://github.com/sn4k3/UVtools/files/7794092/Layer.7.zip)
[Layer 8.zip](https://github.com/sn4k3/UVtools/files/7794099/Layer.8.zip)
.

sn4k3, 2021-12-31:
Videos and files must be uploaded somewhere else and attach here, like wetransfer or any cloud.
With that resolution i was unable to see what times are exposing, but layer 7 first tile show for too long, looks like bottom exposure, this may due the firmware transition layers, what you are experiencing is a fade time from 25s to current time, so you will get 24s on that tile and it will go down as layers transition. You must have a thicker base to over pass this problem. You have a 0.3mm base which is too low, my defaults are ok, if you use 1mm instead the transition layers will be disabled when first tile comes. Try it...
The only way to control transition layers is with version 516 and set them to 0 (It should disable the enforced transition layers)

SaberOfBorg, 2021-12-31:
Tiago:
Thanks for looking-into this.  
Bottom exposure time was set to 25sec . . . as I watch the Layer 7 video I see that the tiles are exposed at: 25, 20, 18, 16....4 sec.  I guess I don't fully understand the purpose of these "transition layers", but, I presume it has something to do with making the 1st tile of layer 8 (which I'd presumed was the way all the subsequent layers were being printed) 4sec instead of the intended 1.5sec . . . but, it seems strange that all of the other tiles appeared to just be 'shifted' 1 position over.

I will take your advice and re-generate the PWMX file with a 1.0mm base (20 layers instead of 6) and see what happens, especially with the 1st tile of the actual test layers (which should be 1.5sec).

I'll let you know what I see . . . thanks again for your help.

SaberOfBorg, 2021-12-31:
Tiago:
I've generated a new test file based on your recommendation . . . please review and comment
[Exposure Test Anycubic Grey.zip](https://github.com/sn4k3/UVtools/files/7796698/Exposure.Test.Anycubic.Grey.zip)
.

sn4k3, 2021-12-31:
It's not ok... You use too much bottom layers (20), also afecting tiles!
Use 3-4 bottom layers

[image]

SaberOfBorg, 2021-12-31:
OK . . . when you said my "base" was too low (0.3mm) and to use 1.0mm I thought you were talking about "bottom layers".  Sorry I misunderstood "base" for "bottom layers". I will read this more thoroughly later today and try again . . .

SaberOfBorg, 2021-12-31:
OK . . . here we go :-)
In Photon Workshop I set the Slice Parameters as per your screen-shot:
[image]

And then sliced it to a test file in PWMX format:
[image]
[image]

I loaded it into UV Tools, and then went-into the Exposure Time Finder tool:.
Bottom Layers set to 3, starting (lowest) exposure time set to 1.5sec:
[image]

I have NEVER adjusted "Base Height":
[image]

Hopefully this file passes your thorough inspection :-)

[Exposure Test Anycubic Grey.zip](https://github.com/sn4k3/UVtools/files/7796930/Exposure.Test.Anycubic.Grey.zip)

sn4k3, 2021-12-31:
It looks ok now, if transition layer count = 0 really works, you should expect layer 3 in uvtools, 4 in printer, to expouse at 1.5s.
But this version (516) requires a new firmware, i dont know if your printer already have it... If it fails to start you know xD

SaberOfBorg, 2021-12-31:
When you say "this version (516)" what are you referring to?  What I'm using is:
Photon Workshop: V2.1.29  RC12
UV Tools: V2.27.4
Photon Mono X Firmware: V3.5.4 (latest from their web-site)

These test results were really strange . . .
Layers 1, 2, 3: 20sec full-screen (14 test strips)
Layer 4: 20.0, 16.3, 14.6, 13.0, 11.2, 9.6, 8.0, 6.4, 4.6, 3.2, 3.0, 3.5, 3.8, 4.0 sec
Layers 5, 6, 7: 4.2, 1.7, 1.9, 2.0, 2.25, 2.35, 2.4, 2.7, 2.8, 3.1, 3.35, 3.5, 3.7, 4.0 sec  (avg of 3 layers)

I was better-off with my 1st attempt where the majority of the layers (the top most) were being printed in a steadily-increasing order, but, with them shifted by 1 test-slot from where they should have been. Guess I'll go back to using the Anycubic-supplied "R_E_R_F" test file.

:-(

sn4k3, 2021-12-31:
Ok so your printer does not respect the 0 transition layers, you still have them.
Your firmware does not know the version 516 of the file format.

So what you can do to by pass them, use this option:
[image]

SaberOfBorg, 2022-01-01:
OK . . thanks for the advice.
I've regenerated the file (see attached) per your recommendation.
At my next opportunity I will "play" the file to see how it exposes things and let you know how it turned-out.

For what it's worth, I've adjusted my exposure list (was 1.5s to 4.1s) because I've since run the Anycubic-provided R_E_R_F exposure test and it appeared that for the Grey resin I'm using the best exposure was about 1.2s to 1.3s.  So, I've now revised the UV Tools exposure test to only 7 tiles from 1.0s to 1.6s in 0.1s increments.
[Exposure Test Anycubic Grey.zip](https://github.com/sn4k3/UVtools/files/7798186/Exposure.Test.Anycubic.Grey.zip)

In case you are interested, prior to printing anything I preheat my resin to about 26`C to 28`C by placing the bottle in a sink full of hot water for about 30minutes.  I have also attached two 8"x12" 16W reptile heaters to the sides of my printer's yellow cover, and surrounded the 3 sides (and top) with mirrored foam insulation, and preheat everything (and run the print) with the temperature controller set to 30`C (and the probe adjacent to the Resin Vat).
[image]
[image]

Cheers.

sn4k3, 2022-02-14:
It seens if you set BottomLayerCount to 0 before generate the test, this will be able to print, can you verify?

closed 2022-03-13 by sn4k3