sn4k3/UVtools #396: [BUG]

g3gg0, 2022-01-06:
First of all: great tool, thanks :)
Some previous version I used a month ago, worked without any issues.
Not sure if that is due to the update i just downloaded or because of the input format.
I did not change the slicer version, just a newer UVTools.

## System
 - OS: Windows 10 Version 10.0.19044.1415
 - Slicer: Photon Workshop v2.1.26RC0
 - Printer: Anycubic Photon S
 - UVtools version: 2.27.5

UVtools v2.27.5
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Processor cores: 12
Memory RAM: 15.23 / 31.94 GB
Runtime: win10-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.11
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1200 @ 100% (Primary) (On this)
    WA: 1920 x 1160    UA: 1920 x 1160
2: 1920 x 1200 @ 100%
    WA: 1920 x 1160    UA: 1920 x 1160

Path: C:\Program Files\UVtools

## Describe the bug
Opening and saving a file causes the printer to show no icon for the saved file.
When trying to print that file, it says "File format error"

## To Reproduce
Steps to reproduce the behavior:
1. Open a sliced file
2. Save to a new file
3. Try to print
4. See error

## Screenshots
https://youtu.be/PQuE2Fv0euw

## Files
(cannot attach the files, even as .zip)
Slicer original: https://1drv.ms/u/s!Amx_Fd-zOoXnnswes2bb3AbyxgHsmw?e=kiWGzk
UVTools output: https://1drv.ms/u/s!Amx_Fd-zOoXnnswdrA2Tk5xv_AZrRw?e=VQw6sm
Delta: 
[image]

sn4k3, 2022-01-06:
Bug is on partial/soft save. Full encode do not have this problem.
Will fix on next patch

g3gg0, 2022-01-06:
Can confirm that "Convert to" works - if that is a "full encode" :)
Thanks for pointing out!

sn4k3, 2022-01-06:
It is, you can simply go to pixel editor, erase a black pixel and apply. That will trigger a full encode

sn4k3, 2022-01-07:
Fixed

closed 2022-01-07 by sn4k3