sn4k3/UVtools #399: [BUG] Inverted brush operation in "Pixel editor"

g3gg0, 2022-01-06:
Sorry to open another issue :)

## System
 - OS: Windows 10 Version 10.0.19044.1415
 - Slicer: Photon Workshop v2.1.26RC0
 - Printer: Anycubic Photon S
 - UVtools version: 2.27.5

UVtools v2.27.5
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Processor cores: 12
Memory RAM: 15.23 / 31.94 GB
Runtime: win10-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.11
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1200 @ 100% (Primary) (On this)
    WA: 1920 x 1160    UA: 1920 x 1160
2: 1920 x 1200 @ 100%
    WA: 1920 x 1160    UA: 1920 x 1160

Path: C:\Program Files\UVtools

## Describe the bug
When using Drawing tools, rotation angle is inverted between cursor and preview/operation

## To Reproduce
Steps to reproduce the behavior:
1. Open a file
2. Click on 'Pixel Editor'
3. Select "Line" shape with huge diameter (e.g. 2560)
4. Set rotation to e.g. 30 degree
5. Cursor shows which pixels will get set/cleared
6. Set or clear some pixels (note: the tooltip says "Hold CTRL to clear pixels", but instead you have to right click)
7. The operation preview will show the operation with an inverted angle

https://www.youtube.com/watch?v=cH9jzrV-Grg

## Expected behavior
preview cursor and operation should match

## Screenshots
[image]
[image]

sn4k3, 2022-01-06:
Yes line seens incorrect, while other shapes are ok...

sn4k3, 2022-01-07:
Fixed

closed 2022-01-07 by sn4k3