sn4k3/UVtools #408: [BUG] UVtools crashes when switching to issues tab

10maurycy10, 2022-01-19:
## System
 - OS: x68_64 arch linux
 - Slicer: none
 - Printer:  none
 - UVtools version: tested on 2.27.4 and 2.27.6

## Reproduction steps
1. Run UVtools (binary, ``dotnet UVtools.dll``, ``./UVtools.sh``)
2. Click on the issues tab.
3. observe crash

## Stderr output
```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.X11.X11Window.SetTransparencyLevelHint(WindowTransparencyLevel transparencyLevel) in /_/src/Avalonia.X11/X11Window.cs:line 1160
   at Avalonia.Controls.TopLevel.<>c.<.cctor>b__16_0(TopLevel tl, AvaloniaPropertyChangedEventArgs e) in /_/src/Avalonia.Controls/TopLevel.cs:line 101
   at System.Reactive.Subjects.Subject`1.OnNext(T value) in /_/Rx.NET/Source/src/System.Reactive/Subjects/Subject.cs:line 145
   at Avalonia.AvaloniaObject.RaisePropertyChanged[T](AvaloniaPropertyChangedEventArgs`1 change) in /_/src/Avalonia.Base/AvaloniaObject.cs:line 756
   at Avalonia.AvaloniaObject.Avalonia.PropertyStore.IValueSink.ValueChanged[T](AvaloniaPropertyChangedEventArgs`1 change) in /_/src/Avalonia.Base/AvaloniaObject.cs:line 531
   at Avalonia.PropertyStore.ConstantValueEntry`1.RaiseValueChanged(IValueSink sink, IAvaloniaObject owner, AvaloniaProperty property, Optional`1 oldValue, Optional`1 newValue) in /_/src/Avalonia.Base/PropertyStore/ConstantValueEntry.cs:line 75
   at Avalonia.ValueStore.BatchUpdate.End() in /_/src/Avalonia.Base/ValueStore.cs:line 454
   at Avalonia.ValueStore.EndBatchUpdate() in /_/src/Avalonia.Base/ValueStore.cs:line 47
   at Avalonia.StyledElement.DetachStyles() in /_/src/Avalonia.Styling/StyledElement.cs:line 769
   at Avalonia.StyledElement.OnDetachedFromLogicalTreeCore(LogicalTreeAttachmentEventArgs e) in /_/src/Avalonia.Styling/StyledElement.cs:line 694
   at Avalonia.StyledElement.Avalonia.LogicalTree.ILogical.NotifyDetachedFromLogicalTree(LogicalTreeAttachmentEventArgs e) in /_/src/Avalonia.Styling/StyledElement.cs:line 389
   at Avalonia.Controls.TopLevel.HandleClosed() in /_/src/Avalonia.Controls/TopLevel.cs:line 368
   at Avalonia.Controls.WindowBase.HandleClosed() in /_/src/Avalonia.Controls/WindowBase.cs:line 202
   at Avalonia.X11.X11Window.Cleanup() in /_/src/Avalonia.X11/X11Window.cs:line 813
   at Avalonia.X11.X11Window.Dispose() in /_/src/Avalonia.X11/X11Window.cs:line 783
   at Avalonia.Controls.Primitives.PopupRoot.Dispose() in /_/src/Avalonia.Controls/Primitives/PopupRoot.cs:line 77
   at Avalonia.Controls.ToolTip.Close() in /_/src/Avalonia.Controls/ToolTip.cs:line 290
   at Avalonia.Controls.ToolTip.IsOpenChanged(AvaloniaPropertyChangedEventArgs e) in /_/src/Avalonia.Controls/ToolTip.cs:line 237
   at System.Reactive.Subjects.Subject`1.OnNext(T value) in /_/Rx.NET/Source/src/System.Reactive/Subjects/Subject.cs:line 145
   at Avalonia.AvaloniaObject.RaisePropertyChanged[T](AvaloniaPropertyChangedEventArgs`1 change) in /_/src/Avalonia.Base/AvaloniaObject.cs:line 756
   at Avalonia.AvaloniaObject.Avalonia.PropertyStore.IValueSink.ValueChanged[T](AvaloniaPropertyChangedEventArgs`1 change) in /_/src/Avalonia.Base/AvaloniaObject.cs:line 544
   at Avalonia.ValueStore.NotifyValueChanged[T](AvaloniaProperty`1 property, Optional`1 oldValue, BindingValue`1 newValue, BindingPriority priority) in /_/src/Avalonia.Base/ValueStore.cs:line 366
   at Avalonia.ValueStore.SetExisting[T](Object slot, StyledPropertyBase`1 property, T value, BindingPriority priority) in /_/src/Avalonia.Base/ValueStore.cs:line 266
   at Avalonia.ValueStore.SetValue[T](StyledPropertyBase`1 property, T value, BindingPriority priority) in /_/src/Avalonia.Base/ValueStore.cs:line 100
   at Avalonia.Controls.ToolTip.SetIsOpen(Control element, Boolean value) in /_/src/Avalonia.Controls/ToolTip.cs:line 121
   at Avalonia.Controls.ToolTipService.Close(Control control) in /_/src/Avalonia.Controls/ToolTipService.cs:line 138
   at Avalonia.Controls.ToolTipService.ControlPointerLeave(Object sender, PointerEventArgs e) in /_/src/Avalonia.Controls/ToolTipService.cs:line 107
   at Avalonia.Interactivity.EventRoute.RaiseEventImpl(RoutedEventArgs e) in /_/src/Avalonia.Interactivity/EventRoute.cs:line 152
   at Avalonia.Interactivity.EventRoute.RaiseEvent(IInteractive source, RoutedEventArgs e) in /_/src/Avalonia.Interactivity/EventRoute.cs:line 79
   at Avalonia.Interactivity.Interactive.RaiseEvent(RoutedEventArgs e) in /_/src/Avalonia.Interactivity/Interactive.cs:line 123
   at Avalonia.Input.MouseDevice.ClearPointerOver(IPointerDevice device, UInt64 timestamp, IInputRoot root, PointerPointProperties properties, KeyModifiers inputModifiers) in /_/src/Avalonia.Input/MouseDevice.cs:line 383
   at Avalonia.Input.MouseDevice.LeaveWindow(IMouseDevice device, UInt64 timestamp, IInputRoot root, PointerPointProperties properties, KeyModifiers inputModifiers) in /_/src/Avalonia.Input/MouseDevice.cs:line 193
   at Avalonia.Input.MouseDevice.ProcessRawEvent(RawPointerEventArgs e) in /_/src/Avalonia.Input/MouseDevice.cs:line 142
   at Avalonia.Input.InputManager.ProcessInput(RawInputEventArgs e) in /_/src/Avalonia.Input/InputManager.cs:line 37
   at Avalonia.X11.X11Window.<ScheduleInput>b__121_0() in /_/src/Avalonia.X11/X11Window.cs:line 743
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 169
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 245
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 132
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 187
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 47
```

sn4k3, 2022-01-19:
duplicate of: https://github.com/sn4k3/UVtools/issues/378

closed 2022-01-19 by sn4k3

10maurycy10, 2022-01-19:
I guess I will use 2.27.0 now. :(

sn4k3, 2022-01-19:
> I guess I will use 2.27.0 now. :(

No other way than wait for a patch from Avalonia