sn4k3/UVtools #410: Segmentaiton fault at startup on Fedora 35 (Latest UVtools)

mcdviii, 2022-01-29:
## System
 - OS: Fedora 35
 - Slicer: 
 - Printer: 
 - UVtools version: UVtools_rhel-x64_v2.27.7

## Describe the bug
After successfully linking libraries & building the `libcvextern.sh` file from `emgucv` (and transferring the file to UVtools root), I get the following errors when trying to run `./UVtools.sh`:

```
./UVtools.sh: line 3: 25854 Segmentation fault      (core dumped) ./UVtools
./UVtools.sh: line 3: 25904 Segmentation fault      (core dumped) dotnet UVtools.dll
```
 
## Exceptions Log file
`sudo journalctl -xe`:
```
Jan 28 20:09:15 [RDCTD] abrt-notification[25382]: [🡕] Process 25242 (dotnet) crashed in hb_shape_plan_create_cached2()
░░ Subject: ABRT has detected unexpected termination: dotnet
░░ Defined-By: ABRT
░░ Support: https://bugzilla.redhat.com/
░░ Documentation: man:abrt(1)
░░
░░ dotnet killed by SIGSEGV
░░
░░ #1 [libharfbuzz.so.0] hb_shape_plan_create_cached2
░░ #2 [libHarfBuzzSharp.so] e1957bdf415362ccca51d175ecd378f883d9db93+719924
░░ #3  (null)+140517932549866
░░ #4  (null)+140517932544487
░░ #5  (null)+140517932466325
░░
░░ Use the abrt command-line tool for further analysis or to report
░░ the problem to the appropriate support site.
```

sn4k3, 2022-01-29:
For reference here is the debug of dependencies:

```
[tiago@fedora BuildInfo.NetCore.Console]$ LD_DEBUG=libs dotnet run
      5376:	find library=libstdc++.so.6 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/libstdc++.so.6
      5376:	
      5376:	find library=libm.so.6 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/libm.so.6
      5376:	
      5376:	find library=libgcc_s.so.1 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/libgcc_s.so.1
      5376:	
      5376:	find library=libc.so.6 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/libc.so.6
      5376:	
      5376:	
      5376:	calling init: /lib64/ld-linux-x86-64.so.2
      5376:	
      5376:	
      5376:	calling init: /lib64/libc.so.6
      5376:	
      5376:	
      5376:	calling init: /lib64/libgcc_s.so.1
      5376:	
      5376:	
      5376:	calling init: /lib64/libm.so.6
      5376:	
      5376:	
      5376:	calling init: /lib64/libstdc++.so.6
      5376:	
      5376:	
      5376:	initialize program: dotnet
      5376:	
      5376:	
      5376:	transferring control: dotnet
      5376:	
      5376:	
      5376:	calling init: /usr/lib64/dotnet/host/fxr/6.0.0/libhostfxr.so
      5376:	
      5376:	
      5376:	calling init: /usr/lib64/dotnet/shared/Microsoft.NETCore.App/6.0.0/libhostpolicy.so
      5376:	
      5376:	find library=libunwind-x86_64.so.8 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/libunwind-x86_64.so.8
      5376:	
      5376:	find library=libunwind.so.8 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/libunwind.so.8
      5376:	
      5376:	
      5376:	calling init: /lib64/libunwind.so.8
      5376:	
      5376:	
      5376:	calling init: /lib64/libunwind-x86_64.so.8
      5376:	
      5376:	
      5376:	calling init: /usr/lib64/dotnet/shared/Microsoft.NETCore.App/6.0.0/libcoreclr.so
      5376:	
      5376:	find library=liblttng-ust.so.0 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
      5376:	  trying file=/lib64/liblttng-ust.so.0
      5376:	
      5376:	find library=liblttng-ust-tracepoint.so.0 [0]; searching
      5376:	 search cache=/etc/ld.so.cache
[... 736 lines cut ...]
      5376:	
      5376:	
      5376:	calling fini: /lib64/libstdc++.so.6 [0]
      5376:	
      5376:	
      5376:	calling fini: /lib64/libgcc_s.so.1 [0]
      5376:	
      5376:	
      5376:	calling fini: /lib64/libm.so.6 [0]
      5376:	
      5376:	
      5376:	calling fini: /usr/lib64/dotnet/shared/Microsoft.NETCore.App/6.0.0/libSystem.Security.Cryptography.Native.OpenSsl.so [0]
      5376:	
      5376:	
      5376:	calling fini: /lib64/libssl.so.1.1 [0]
      5376:	
      5376:	
      5376:	calling fini: /lib64/libcrypto.so.1.1 [0]
      5376:	
      5376:	
      5376:	calling fini: /lib64/libz.so.1 [0]
      5376:	
```

I will latter look into this

sn4k3, 2022-02-01:
To fix this run this command:

`sudo ln -s /usr/lib64/libdl.so.2 /usr/lib64/libdl.so`

and use the UVtools_v***_linux_x64.zip (Dont use the rhle)

bufothefrog, 2022-02-02:
Tested working on my system

mcdviii, 2022-02-02:
Can confirm this works on my system as well. I have not tried anything else, but the application launches without problems. Will close this issue. Thank you!

closed 2022-02-02 by mcdviii