sn4k3/UVtools #411: [BUG] segfault when trying to open a CTB file

nonchip, 2022-01-30:
## System
 - OS: void linux
 - Slicer: any (confirmed chitubox and lychee)
 - Printer: creality LD-002H (not that that matters)
 - UVtools version: 2.27.7

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:
UVtools v2.27.7
Operative system: Linux 5.15.12_1 #1 SMP Wed Dec 29 19:18:57 UTC 2021 X64
Processor: AMD Ryzen 5 2600 Six-Core Processor
Processor cores: 12
Memory RAM: 27.63 / 31.37 GB
Runtime: void-x64
Framework: .NET 5.0.13
AvaloniaUI: 0.10.12
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1200 @ 100% (Primary)
    WA: 1920 x 1168    UA: 1920 x 1168
2: 1680 x 1050 @ 100%
    WA: 1680 x 1050    UA: 1680 x 1050

Path: /home/kyra/bin/UVtools_linux-x64_v2.27.7

General configuration for OpenCV 4.5.5 =====================================
  Version control:               4.5.5-237-gaa8b5d2f70

  Extra modules:
    Location (extra):            /var/lib/buildbot/workers/worker/cv_ubuntu/build/opencv_contrib/modules
    Version control (extra):     4.5.5-62-g7cb6b4d5

  Platform:
    Timestamp:                   2022-01-04T20:38:52Z
    Host:                        Linux 5.10.60.1-microsoft-standard-WSL2 x86_64
    CMake:                       3.16.3
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3
    Dispatched code generation:  SSE4_1 SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_1 (16 files):         + SSSE3 SSE4_1
      SSE4_2 (1 files):          + SSSE3 SSE4_1 POPCNT SSE4_2
      FP16 (0 files):            + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 AVX
      AVX (4 files):             + SSSE3 SSE4_1 POPCNT SSE4_2 AVX
      AVX2 (31 files):           + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (5 files):      + SSSE3 SSE4_1 POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /usr/bin/c++  (ver 9.3.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections  -msse -msse2 -msse3 -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--exclude-libs,libippicv.a -Wl,--exclude-libs,libippiw.a   -Wl,--gc-sections -Wl,--as-needed  
    Linker flags (Debug):        -Wl,--exclude-libs,libippicv.a -Wl,--exclude-libs,libippiw.a   -Wl,--gc-sections -Wl,--as-needed  
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          /var/lib/buildbot/workers/worker/cv_ubuntu/build/platforms/ubuntu/20.04/build/install/lib/libhdf5.a vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy Iconv::Iconv dl m pthread rt
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc ippiw ippicv
[... 219 lines cut ...]
#25 0x00007fff22d1e160 in ?? ()
#26 0x00007fff100016b0 in ?? ()
#27 0x00007fff10003400 in ?? ()
#28 0x00007fff10003480 in ?? ()
#29 0x00007fff100016f0 in ?? ()
#30 0x000000004bdc7280 in ?? ()
#31 0x00007fff4bdc7388 in ?? ()
#32 0x00007fff4bdc7360 in ?? ()
#33 0x00007fff4bdc7338 in ?? ()
#34 0x00007fff4bdc7310 in ?? ()
#35 0x0000000400000004 in ?? ()
#36 0x00007fff4bdc72e0 in ?? ()
#37 0x00007fff4bdc72b0 in ?? ()
#38 0x00007fff4bdc7280 in ?? ()
#39 0x00007fff4bdc71d8 in ?? ()
#40 0x00007fff22d1e600 in ?? ()
#41 0x00007fff82298813 in ?? ()
#42 0x00000000ffffffff in ?? ()
#43 0x0000000000000002 in ?? ()
#44 0x00007fff8203b788 in ?? ()
#45 0x00007fff82297a90 in ?? ()
#46 0x00007fff22d1e1f0 in ?? ()
#47 0x00007fff8203b788 in ?? ()
#48 0x0000000000000000 in ?? ()
```

sn4k3, 2022-01-30:
Who says open bug issues aren't fun? 😂
This seen a extension of https://github.com/sn4k3/UVtools/issues/378
Please try to downgrade to v2.27.0 and see if problem happens

tescalada, 2022-02-02:
Downgrading to v2.27.0 fixed the segfault when opening CTB files for me.

closed 2022-02-14 by sn4k3