sn4k3/UVtools #416: 2.27.7 Crashing on Detection - M1 Mac

ok1ok1, 2022-02-11:
Application will crash whenever running detection. If problem report dialog appears the application can be relaunched from the dialog and the application will function normally. Unfortunately the dialog does not always appear after a crash. The app will only function normally if launched from the problem report dialog.

## System
 - OS: MacOS 11.2.2 (M1 Mac)
 - Slicer: ANY
 - Printer: ANY
 - UVtools version: 2.27.7

A problem report & demonstration video is attached
[Crash Report.txt](https://github.com/sn4k3/UVtools/files/8048841/Crash.Report.txt)

https://user-images.githubusercontent.com/5963693/153597329-90c5a0b4-ce2c-4ec5-94e3-f34ee9cd9a6f.mp4

d

sn4k3, 2022-02-11:
I guess this problem is related to OpenCV this time...
Try to uncheck Islands on detection, does it still crash?

ok1ok1, 2022-02-11:
If islands is unchecked it does not crash.

sn4k3, 2022-02-11:
> problem report dialog

What do you mean? Is that some MacOS stuff?

ok1ok1, 2022-02-11:
It probably is, I'll try to explain better. 

When the app initially crashes often times a problem report window opens. It gives the option to reopen the application. 
So long as the app is launched using the reopen button the problem is resolved. The application runs normally, even with island detection enabled.

If the app is not launched using this method it will continue to crash. 

I've attached a screenshot of the problem report window. 
[image]

sn4k3, 2022-02-11:
That's strange... I dont know if that "Reopen" lunch in a different way but looks like it.
If you change tab to Log, go back to issues and detect does it still crash?

sn4k3, 2022-02-11:
Ok, so problem seens on OpenCV on a function.
It will not happen if you downgrade UVtools wish use a pior OpenCV version
It also happen on Linux but not Windows

> Thread 19:
> 0 libcvextern.dylib 0x00000001ae73f097 cv::connectedcomponents::LabelingBolelli4CParallel<int, unsigned char, cv::connectedcomponents::CCStatsOp>::FirstScan::operator()(cv::Range const&) const + 839
> 1 libcvextern.dylib 0x00000001ae514569 cv::parallel_for_(cv::Range const&, cv::ParallelLoopBody const&, double) + 393
> 2 libcvextern.dylib 0x00000001ae72b183 cv::connectedcomponents::LabelingBolelli4CParallel<int, unsigned char, cv::connectedcomponents::CCStatsOp>::operator()(cv::Mat const&, cv::Mat&, int, cv::connectedcomponents::CCStatsOp&) + 435
> 3 libcvextern.dylib 0x00000001ae6df372 int cv::connectedComponents_sub1cv::connectedcomponents::CCStatsOp(cv::Mat const&, cv::Mat&, int, int, cv::connectedcomponents::CCStatsOp&) + 626
> 4 libcvextern.dylib 0x00000001ae6defe6 cv::connectedComponentsWithStats(cv::_InputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, cv::_OutputArray const&, int, int, int) + 438

ok1ok1, 2022-02-12:
Initially it was a big problem but the work around is fairly simple. I can deal with it until a solution is found. Thanks for looking into it. You've created an amazing tool. It has become an essential part of my workflow.

antwal, 2022-02-12:
Same problem on all files, working version 2.27.6

sn4k3, 2022-02-12:
To use the latest UVtools version you can replace the `libcvextern.dylib` from the working version into the new. It will work without the problem

ok1ok1, 2022-02-13:
No problems with 2.28.0. The issue appears to be resolved. 
Thank you!

antwal, 2022-02-13:
it works, only the process is still very slow compared to previous versions.

sn4k3, 2022-02-14:
> it works, only the process is still very slow compared to previous versions.

But it uses the OpenCV from previous version...
Current problem: https://github.com/opencv/opencv/issues/21366

closed 2022-02-14 by sn4k3