sn4k3/UVtools #424: Exposure time finder window opens off screen

nikdfish, 2022-02-26:
If display resolution is set higher than 1600 x 900, UVtools the Exposure time finder windows opens off screen.

Example:  open Uvtools, click on "Calibration",  select "Exposure time finder", 

Expected result is a new window, on screen, for  Exposure time finder

Observed result is Uvtools main window is non responsive and Exposure time finder window is not visible.

User can use the Windows taskbar to left click select the Uvtools task icon, then right click select Exposure time finder window icon under Uvtools and use either "move" or"maximize" right click menu options to cause Exposure time finder window to appear on the display.

If the user selects a screen resolution of 1600 x 900 or less, the Exposure time finder window will open on the display as expected.

===================

UVtools v2.29.0
Operative system: Microsoft Windows 10.0.22000 X64
Processor: Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
Processor cores: 12
Memory RAM: 8.65 / 31.86 GB
Runtime: win10-x64
Framework: .NET 5.0.14
AvaloniaUI: 0.10.12
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1032    UA: 1920 x 1032

Path: C:\Program Files\UVtools

General configuration for OpenCV 4.5.5 =====================================
  Version control:               4.5.5-237-gaa8b5d2f70

  Extra modules:
    Location (extra):            D:/bb/cv_x64/build/build_x64/../opencv_contrib/modules
    Version control (extra):     4.5.5-62-g7cb6b4d5

  Platform:
    Timestamp:                   2022-01-05T15:01:30Z
    Host:                        Windows 10.0.22000 AMD64
    CMake:                       3.21.3
    CMake generator:             Visual Studio 16 2019
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
    MSVC:                        1929
    Configuration:               Debug Release MinSizeRel RelWithDebInfo

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe  (ver 19.29.30138.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MD /O2 /Ob2 /DNDEBUG 
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MDd /Zi /Ob0 /Od /RTC1 
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP   /MD /O2 /Ob2 /DNDEBUG 
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP /MDd /Zi /Ob0 /Od /RTC1 
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO 
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL 
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          wsock32 comctl32 gdi32 ole32 setupapi ws2_32 D:/bb/cv_x64/build/build_x64/install/lib/freetype.lib D:/bb/cv_x64/build/build_x64/install/lib/harfbuzz.lib D:/bb/cv_x64/build/build_x64/install/lib/libhdf5.lib vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc

  OpenCV modules:
    To be built:                 alphamat aruco barcode bgsegm bioinspired calib3d ccalib core datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hdf hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking ts video videoio videostab viz wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    python_bindings_generator python_tests world
    Disabled by dependency:      -
[... 146 lines cut ...]
085: Newtonsoft.Json, Version=13.0.0.0
086: System.Diagnostics.Tracing, Version=5.0.0.0
087: System.Diagnostics.DiagnosticSource, Version=5.0.0.0
088: System.Net.Security, Version=5.0.0.0
089: System.Security.Cryptography.X509Certificates, Version=5.0.0.0
090: BinarySerializer, Version=8.6.2.2
091: System.Threading.Tasks, Version=5.0.0.0
092: System.Text.Encoding, Version=5.0.0.0
093: System.IO, Version=5.0.0.0
094: System.Reflection.TypeExtensions, Version=5.0.0.0
095: System.Reflection, Version=5.0.0.0
096: System.Reflection.Extensions, Version=5.0.0.0
097: System.Net.Sockets, Version=5.0.0.0
098: System.Net.NameResolution, Version=5.0.0.0
099: System.IO.Compression.ZipFile, Version=5.0.0.0
100: System.Collections.Specialized, Version=5.0.0.0
101: System.Threading.Tasks.Parallel, Version=5.0.0.0
102: System.Security.Cryptography.Primitives, Version=5.0.0.0
103: System.Security.Principal.Windows, Version=6.0.0.0
104: System.Security.Claims, Version=5.0.0.0
105: System.Security.Principal, Version=5.0.0.0
106: System.Security.Cryptography.Encoding, Version=5.0.0.0
107: System.Runtime.Numerics, Version=5.0.0.0
108: System.IO.FileSystem.DriveInfo, Version=5.0.0.0
109: Microsoft.Win32.Registry, Version=6.0.0.0

sn4k3, 2022-02-26:
If you often change resolution that sometimes can mess with program windows positions, some keep last position, meaning if you change resoltuion next time window open will be on wrong position. It happen with other applications as well, eg: notepad for me.
You can try to reboot UVtools on new resolution and move window to correct position, then quit and try again, does window still spawn in wrong position?

nikdfish, 2022-02-26:
I tried closing UVtools after moving the window to a more proper location, but after restarting UVtools the tool window still opens off screen.   

Other tool windows do show the same behaviour.  Just verified with the "settings" window.   

I normally don't change resolution, but was trying different ones to see if it made a difference in this situation.  All resolutions at or below 1600x900 see the tool windows opening on the display.

My memory may be betraying me, but I don't recall this behaviour when I used an earlier version of UVtools.  

FWIW, I am using a Dell laptop (G3 3779)  with a HDMI connected cloned display.   Both laptop & external display have the same resolution (1920x1080).  The display adapter is a  NVIDIA GeForce GTX 1050 Ti

sn4k3, 2022-02-26:
Try to disconnect HDMI from port and use only laptop screen
Reopen UVtools
Open a window, Does it show same problem?

ToolWindows are positioned at your screen center, resolution size come from UI framework.

The about window shows you the current resolution your screen report

```
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1032    UA: 1920 x 1032
```

Then tool window is placed at center of X and top of Y.
Something is messing with the return size from framework

nikdfish, 2022-02-26:
With the HDMI cable disconnected I got the same behavior.  I also tried closing UVtools after repositioning the tool window then reopening.  The tool window still ended up opening off screen.

Screen shots follow:

screen shot, HDMI disconnected, UVtools and tool window open but tool window off screen (see task bar)
[image]

screen shot, HDMI disconnected, UVtools and tool window open after using task bar right click menu to move tool window onto visible screen location.
[image]

just for reference, when the right click "move" option is selected, the mouse cursor goes to an upper extreme left position on the screen (half the cross is off screen).  the cursor was then moved back to screen center & an arrow key pressed to get the tool window to appear.  

(and yes I know I have a rediculously crowded desktop  - I tend just find things better via position rather than list hierarchy)

nikdfish, 2022-02-26:
55, 55

sn4k3, 2022-02-27:
Ok then try this:

1. Resize main window a bit (Direction doesnt matter)
2. Move main window a bit with the mouse (Position dont matter)
3. Open a tool window

What the outcome?

nikdfish, 2022-02-27:
tool window appears on-screen, apparently centered left/right on the main UI window & down by  one text line (aprx)

nikdfish, 2022-02-27:
moving by itself has no effect.  resizing by itself will result in the tool window appearing on-screen

sn4k3, 2022-02-27:
Ok so is a bug on the UI framework, it's not updating the real position before you resize it, or maybe app that attach to windows are causing that.
Anyway i will patch to limit and not allow negative values, on my case i have -8,-8 you should have a large value!

Try this test build: https://we.tl/t-DipD9rb23o
Does it fix the problem?

Also without resize main window, on terminal window try to send: `Position` and report back the value. Now it will report the position from Main window.

nikdfish, 2022-02-27:
Running the unzipped package of files, there is no change in behaviour.  The tool window is off screen.
Running "Position" from a terminal window (without moving the main window) returned 52, 52

sn4k3, 2022-02-27:
That's strange.
Try this build: https://we.tl/t-nucDI6KdtE

nikdfish, 2022-02-27:
The second zipped set opens tool windows on-screen & fully visible except for a portion hidden by the task bar & screen bottom.  They appear to be one cascade step down from the main window and  pretty much on the left edge of the screen.

The "Position" command in a terminal window returned  156, 156

sn4k3, 2022-02-27:
Screenshot

nikdfish, 2022-02-27:
UVTools just opened - not moved, not resized
[image]

Exposure time finder window
[image]

Position
[image]

sn4k3, 2022-02-27:
Ok now looks correct given the code.
You need to work with UVtools maximized for the window goes to top

The code which set tool window position is:

```
Position = new PixelPoint(
  Math.Max(0, (int)(Math.Max(0, App.MainWindow.Position.X) + App.MainWindow.Width / 2 - Width / 2)),
  Math.Max(0, App.MainWindow.Position.Y) + 20
);
```

It's all dependent on MainWindow position

nikdfish, 2022-02-27:
Thanks for your good work!

nikdfish, 2022-02-28:
my issue has been resolved

closed 2022-02-28 by nikdfish