sn4k3/UVtools #426: [BUG]

bljfour, 2022-03-10:
## System
 - OS: MacOS 12.2.1
 - Slicer: Photon Workshop 2.1.29rc29
 - Printer: Anycubic Photon Ultra
 - UVtools version: 2.29.0

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:
UVtools v2.29.0
Operative system: Darwin 21.3.0 Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_ARM64_T6000 X64
Processor: Apple M1 Pro
Processor cores: 10
Memory RAM: 15.89 / 30.52 GB
Runtime: osx.11.1-x64
Framework: .NET 5.0.14
AvaloniaUI: 0.10.12
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 3008 x 1692 @ 200% (Primary)
    WA: 3008 x 1582    UA: 1504 x 791
2: 3008 x 1692 @ 200%
    WA: 3008 x 1667    UA: 1504 x 834

Path: /Applications/UVtools.app/Contents/MacOS

General configuration for OpenCV 4.5.4 =====================================
  Version control:               4.5.4-235-g27567928d7

  Extra modules:
    Location (extra):            /Users/tiago/Desktop/emgucv/opencv_contrib/modules
    Version control (extra):     4.5.4-61-gc33e5b60

  Platform:
    Timestamp:                   2021-11-01T19:19:02Z
    Host:                        Darwin 18.7.0 x86_64
    CMake:                       3.21.4
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3 SSSE3 SSE4_1
      requested:                 DETECT
    Dispatched code generation:  SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_2 (1 files):          + POPCNT SSE4_2
      FP16 (0 files):            + POPCNT SSE4_2 FP16 AVX
      AVX (4 files):             + POPCNT SSE4_2 AVX
      AVX2 (30 files):           + POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (5 files):      + POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /Library/Developer/CommandLineTools/usr/bin/c++  (ver 10.0.1.10010046)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /Library/Developer/CommandLineTools/usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,-dead_strip  
    Linker flags (Debug):        -Wl,-dead_strip  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          -framework OpenCL -framework Cocoa -framework AppKit /Users/tiago/Desktop/emgucv/platforms/macos/build/install/lib/libhdf5.a Iconv::Iconv
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libpng libtiff libopenjp2 IlmImf zlib quirc

[... 168 lines cut ...]

```

## Describe the bug
An exception is thrown when trying to generate an exposure test for the Anycubic Photon Ultra.

## To Reproduce
Steps to reproduce the behavior:
1. Load test print included with Anycubic Photon Ultra ("TEST.dlp")
2. Select Calibration->Exposure Time Finder
3. Don't change any settings
4. Press "Exposure time finder" button on the bottom of the window

## Expected behavior

Expected to have a file generated, instead I get this exception:

System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.LayerManager.GetBoundingRectangle(OperationProgress progress)
   at UVtools.Core.LayerManager.get_BoundingRectangle() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Layers\LayerManager.cs:line 178
   at UVtools.Core.FileFormats.FileFormat.get_BoundingRectangle() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 1112
   at UVtools.Core.Operations.Operation..ctor(FileFormat slicerFile) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\Operation.cs:line 300
   at UVtools.Core.Operations.OperationCalibrateExposureFinder.ExecuteInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationCalibrateExposureFinder.cs:line 2250
   at UVtools.Core.Operations.Operation.Execute(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\Operation.cs:line 549
   at UVtools.WPF.MainWindow.<>c__DisplayClass116_0.<RunOperation>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 2364

sn4k3, 2022-03-10:
That's strange. Can you attach the test file here?

bljfour, 2022-03-10:
[TEST.zip](https://github.com/sn4k3/UVtools/files/8220200/TEST.zip)

sn4k3, 2022-03-10:
I'm getting diferent error. Which settings you used inside the tool?
BTW: Anycubic machines are not compatible with multiple exposures

bljfour, 2022-03-10:
I didn’t change any settings in the tool—just loaded the test file, opened
up the exposure finder and clicked the button to generate. I was just
trying to create a default exposure test before changing settings to
generate others.

Besides the Photon Ultra, I have a Photon Mono X and so I was just
following my previous workflow of creating 5 or so different exposure tests
to print and see which settings work best with the resin I have.

sn4k3, 2022-03-10:
I found the problem...
The stock test is taller than your printer resolution, and when it try to fit or move it triggers the error.
If you remove the Bulleye to make it short it will work fine.
That printer is really small :(

[image]

bljfour, 2022-03-10:
Thanks for finding the problem and responding so quickly! And yes, the print area is tiny!

bljfour, 2022-03-10:
And I also see no errors now when I remove the bullseye

sn4k3, 2022-03-10:
If you set Top/bottom margin to 0mm you will be able to fit the whole test
[image]

closed 2022-03-10 by bljfour