sn4k3/UVtools #428: [BUG]

pahofmann, 2022-03-10:
## System
 - OS: Manjaro KDE
 - UVtools version:  latest (2.29)

## Describe the bug
I have trouble getting it to work on the latest Manjaro KDE, using the arch or linux version with all mentioned dependencies installed.
`~/D/UVtools_arch-x64_v2.29.0 ❯❯❯ ./UVtools
Unhandled exception. System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'dl' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libdl: cannot open shared object file: No such file or directory
   at Emgu.Util.Toolbox.Dlopen(String dllname, Int32 mode)
   at Emgu.Util.Toolbox.LoadLibrary(String dllname)
   at Emgu.CV.CvInvoke.LoadUnmanagedModules(String loadDirectory, String[] unmanagedModules)
   at Emgu.CV.CvInvoke.DefaultLoadUnmanagedModules(String[] modules)
   at Emgu.CV.CvInvoke..cctor()
   --- End of inner exception stack trace ---
   at Emgu.CV.CvInvoke.Init()
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\App.axaml.cs:line 66
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run() in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 166
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.X11.X11PlatformThreading.CheckSignaled() in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 164
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 244
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 53
fish: Job 1, './UVtools' terminated by signal SIGABRT (Abort`

## To Reproduce
1. Install dependencies
2. Unpack latest version
3.  run with ./UVtools
4. See error

sn4k3, 2022-03-10:
Try the linux_x64 package instead

pahofmann, 2022-03-10:
Sorry forgot to mention I tried that as well with the same result.

sn4k3, 2022-03-10:
Try this:

`sudo ln -s /lib/x86_64-linux-gnu/libdl.so.2 /lib/x86_64-linux-gnu/libdl.so`

pahofmann, 2022-03-10:
There is no `x86_64-linux-gnu` folder inside `/lib` on any of my manjaro systems, am I missing something?

sn4k3, 2022-03-10:
Can you try to locate the `libdl.so` on your system? What is it path?

printersmate, 2022-03-12:
libdl and pthread have been merged into glibc see:
https://bugs.archlinux.org/task/73830
and no longer exists as separate libs.

sn4k3, 2022-03-13:
> libdl and pthread have been merged into glibc see

That's the package yes, but libdl.so must exist
Mine is located here:

[image]

Check if you have it

pahofmann, 2022-03-13:
I have all the mentioned dependencies installed:
```
openjpeg2 libjpeg-turbo libpng libgeotiff libdc1394 libdc1394 ffmpeg openexr tbb libgdiplus
```

But a `sudo find / -name libdl.so` still returns nothing on both my manjaro systems..

printersmate, 2022-03-13:
For arch based systems, including manjaro, sn4k3's suggestion, modified for system differences,
sudo ln /lib/libdl.so.2 /lib/libdl.so
solves the issue.

(A good nights rest clears the brain fog).

pahofmann, 2022-03-13:
Well I can't link a library that is not there. Find doesn't return lindl.so anywhere.

All the listed dependencies and of course glibc which it should be a part of are installed though.

sn4k3, 2022-03-13:
> Well I can't link a library that is not there. Find doesn't return lindl.so anywhere.
> 
> All the listed dependencies and of course glibc which it should be a part of are installed though.

To find the library you must run this command: `pacman -Ql glibc | grep -E '(libdl|pthread).*so'`

sn4k3, 2022-03-13:
I have created this script to help users.
Not all path's are covered still it tests the most common: `wget -qO - https://raw.githubusercontent.com/sn4k3/UVtools/master/Scripts/libdl-solver.sh | sudo bash`

pahofmann, 2022-03-14:
Thanks for the help! Ugh that was my bad, the find command was of, because it is `libdl.so.2` it didn't turn up.
Pacman lists it correctly:  
```
pacman -Ql glibc | grep -E '(libdl|pthread).*so'
glibc /usr/lib/libdl.so.2
glibc /usr/lib/libpthread.so.0
```
and after `sudo ln -s /usr/lib/libdl.so.2 /usr/lib/libdl.so` it works fine now!

closed 2022-03-14 by pahofmann