sn4k3/UVtools #431: [BUG] UVtools 3.0.0 - arm build not functional

rfield19, 2022-03-12:
## System
 - OS: OSX Monterey 12.3
 - UVtools version: 3.0.0 

## Describe the bug
ARM64 build for osx doesn't run.
Pops a window up with 'The application 'uvtools' can't be opened.'
(non arm64 build of uvtools works fine through rosetta, same as previous versions).

## To Reproduce
Run uvtools on m1 mac
See error :)

## Screenshots
[image]

rfield19, 2022-03-12:
For thoroughness, have just tried it on a second M1 machine - same result unfortunately.

sn4k3, 2022-03-12:
Yes i know. See related here: https://github.com/sn4k3/UVtools/issues/187
You can run it via `sh` command, i dont know yet how to put it working since i have no mac m1 to try.
Need community help on here

rfield19, 2022-03-12:
Ah, ok sorry - I thought 187 was just the previous feature request for the arm build.
I'm 100% happy to test anything / everything on M1 machines you want to throw at me if that helps at all.

Regardless, congrats on releasing v3 though - you must be exhausted!

sn4k3, 2022-03-12:
> Ah, ok sorry - I thought 187 was just the previous feature request for the arm build. 

It is. But problem was also spoted while i send some test builds there.

> I'm 100% happy to test anything / everything on M1 machines you want to throw at me if that helps at all.

I really don't know where to start, i tried multiple things but sending packages for each try is painfull, i would need some remote access to an M1 to accelerate this and dump everything i can try.
One thing you can try is give run permission to all files within `UVtools.app` and also inspect app properties and see permissions

> Regardless, congrats on releasing v3 though - you must be exhausted!

I had it on hold for some days. But not worth to keep it unreleased because of this situation

rfield19, 2022-03-12:
I have an update for you - might help, might make things worse! (let me know if you'd prefer me to post this in 187 instead)...

This is what I get when running 3.0.0 via sh method 

Failed to load /UVtools.app/Contents/MacOS/libhostpolicy.dylib, error: dlopen(/UVtools.app/Contents/MacOS/libhostpolicy.dylib, 0x0001): tried: '/UVtools.app/Contents/MacOS/libhostpolicy.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
An error occurred while loading required library libhostpolicy.dylib from [/UVtools.app/Contents/MacOS/]

sn4k3, 2022-03-12:
> I have an update for you - might help, might make things worse! (let me know if you'd prefer me to post this in 187 instead)...

We can keep this thread as #187 was about the arm request and titled as such.

> This is what I get when running 3.0.0 via sh method
>
> Failed to load /UVtools.app/Contents/MacOS/libhostpolicy.dylib, error: dlopen(/UVtools.app/Contents/MacOS/libhostpolicy.dylib, 0x0001): tried: [...]

I don't know details about @aaaldo managed to run it via sh, but that error is strange, telling it's arm64 but need x64?

rfield19, 2022-03-12:
> 

I thought the same thing - bit of an odd error on an m1 mac.  
Perhaps it doesn't think the compiled binary is arm code?
I have a handy bit of software called SiliconInfo that tells me if it's arm or x64, but only if the software is running...

Anyway, the second m1 I've just tried it on (with a lot less libs installed on it), the sh script doesn't run at all as it can't find dotnet (line 3 of uvtools.sh).

I just installed dotnet via brew and got this fairly horrendous error!
I think it's mostly complaining that libcvextern isn't present in the contents/macos/ dir?

UVtools.app/Contents/MacOS/UVtools.sh: line 3: 46221 Killed: 9               ./UVtools
Unhandled exception. System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'cvextern' or one of its dependencies. In order to help diagnose loading problems, consider setting the DYLD_PRINT_LIBRARIES environment variable: dlopen(libcvextern, 0x0001): tried: 'libcvextern' (no such file), '/usr/local/lib/libcvextern' (no such file), '/usr/lib/libcvextern' (no such file), 'Downloads/UVtools.app/Contents/MacOS/libcvextern' (no such file)
   at Emgu.CV.CvInvoke.RedirectError(CvErrorCallback errorHandler, IntPtr userdata, IntPtr prevUserdata)
   at Emgu.CV.CvInvoke..cctor()
   --- End of inner exception stack trace ---
   at Emgu.CV.CvInvoke.Init()
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\App.axaml.cs:line 198
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run() in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 181
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Native.PlatformThreadingInterface.SignaledCallback.Signaled(Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 39
   at Avalonia.Native.Interop.Impl.__MicroComIAvnSignaledCallbackVTable.Signaled(Void* this, Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/Interop.Generated.cs:line 4337
--- End of stack trace from previous location ---
   at Avalonia.Native.PlatformThreadingInterface.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 90
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 56
UVtools.app/Contents/MacOS/UVtools.sh: line 3: 46222 Abort trap: 6           dotnet UVtools.dll

sn4k3, 2022-03-12:
> I thought the same thing - bit of an odd error on an m1 mac.
> Perhaps it doesn't think the compiled binary is arm code?
> I have a handy bit of software called SiliconInfo that tells me if it's arm or [...]

You dont need install dotnet since it's included on package. But if you do you need the 6.0.3.
If sh fails you better to call binary directly: `./UVtools.app/Contents/MacOS/UVtools`

About libcvextern thats a common error when you missing any dependency.
I suspect fffpmeg need downgrade

```sh
brew install cmake ffmpeg@4
brew link ffmpeg@4
```

Reboot and retry

rfield19, 2022-03-12:
> > I thought the same thing - bit of an odd error on an m1 mac.
> > Perhaps it doesn't think the compiled binary is arm code?
> > I have a handy bit of software called SiliconInfo that tells me if it's arm or x64, but only if the software is running...
> > Anyway, the second m1 I've just tried it on (with a lot less libs installed on it), the sh script doesn't run at all as it can't find dotnet (line 3 of uvtools.sh).
> 
> You dont need install dotnet since it's included on package. But if you do you need the 6.0.3. If sh fails you better to call binary directly: `./UVtools.app/Contents/MacOS/UVtools`
> 
> About libcvextern thats a common error when you missing any dependency. I suspect ffpmeg need downgrade
> 
> ```shell
> brew install cmake ffmpeg@4
> brew link ffmpeg@4
> ```
> 
> Reboot and retry

dotnet definitely wasn't being picked up until I installed it via brew.

Slight improvement though - both machines are now identically producing that longer error message re libcvextern - the libhost error went away after I updated dotnet to latest version...

I'll try downgrading ffmpeg now!!

rfield19, 2022-03-12:
Result...
Downgrading ffmpeg on first machine, and installing ffmpeg on second machine (it didn't have it on there) did the trick...

Just got to figure out why it's only launching via sh now 😅 

(and on benchmark, multithread score has jumped from 180 to 280!)

sn4k3, 2022-03-12:
> Result... Downgrading ffmpeg on first machine, and installing ffmpeg on second machine (it didn't have it on there) did the trick...
> Just got to figure out why it's only launching via sh now 😅

Nice. The stange thing is the emulated version dont require downgrade...

> (and on benchmark, multithread score has jumped from 180 to 280!)

You need to compare version 3.0 x64 with arm, if you comparing prior version it can have slower performance. NET 6.0 optimized some performance

rfield19, 2022-03-12:
> Nice. The stange thing is the emulated version dont require downgrade...
>
>
> You need to compare version 3.0 x64 with arm, if you comparing prior version it can have slower performance. NET 6.0 [...]

I'll try comparing to x64 3.0 ver later this eve.  
Thought this wasn't too shabby for a laptop (the second m1 machine) - it's now beating your i9-9900k :)

[image]

sn4k3, 2022-03-12:
> Thought this wasn't too shabby for a laptop (the second m1 machine) - it's now beating your i9-9900k :)

I guess i need a new CPU 😂
But i hand't update that values since NET 5.0, need to re-run with NET 6.0 too

M1 is a good CPU

aaaldo, 2022-03-12:
> We can keep this thread as #187 was about the arm request and titled as such.
>
>
> I don't know details about @aaaldo managed to run it via sh, but that error is strange, telling it's arm64 but need x64?

hello everybody, 

I ran this command : 

`sh /Applications/UVtools.app/Contents/MacOS/UVtools.sh`

bgyarfas, 2022-03-13:
@sn4k3 `dotnet` is definitely not being found 
```
~/Downloads 
➜ sh UVtools.app/Contents/MacOS/UVtools.sh 
UVtools.app/Contents/MacOS/UVtools.sh: line 3: 75336 Killed: 9               ./UVtools
UVtools.app/Contents/MacOS/UVtools.sh: line 3: dotnet: command not found
```
I just got this M1 a couple weeks ago, haven't even installed homebrew yet. 

`dotnet` doesn't appear to be anywhere in the `.app` either... 
```
find UVtools.app -name "*dotnet*"
```

sn4k3, 2022-03-13:
Again, you dont need dotnet.
instead of `sh` use `bash`

rfield19, 2022-03-13:
@sn4k3 Tiago, I think the confusion maybe is because you're thinking the script line

./UVtools || dotnet UVtools.dll

should be succesfully running ./UVtools, and only if it fails does it run 'dotnet UVtools.dll'.

Unfortunately though, ./UVtools command doesn't run - the process dies instantly, so then it _does_ try to run 'dotnet UVtools.dll'!

Without dotnet being installed via brew in other words, it's not possible to run uvtools.

To put it another way - if you directly try to run 'UVtools.app/Contents/MacOS/UVtools' then it crashes out immediately with a process killed textline, so perhaps it's all a compile issue (may also be why it's necessary to use sh / bash at all, if the main UVtools binary isn't executable?)

*I have naturally also checked UVtools (in the Contents/MacOS/ dir) was chmod +x'ed, just to be sure!

bgyarfas, 2022-03-13:
@sn4k3 Sorry, I misunderstood. I figured that "You don't need install dotnet since it's included on package...." meant there was a standalone `dotnet` somewhere in the app but more likely it's just build against dotnet. I'm downgrading ffmpeg right now and I'll let you know if that works.

bgyarfas, 2022-03-13:
@sn4k3 After installing/downgrading ffmpeg I'm still getting the same error. 
```
~/Downloads 
➜ which ffmpeg  
/opt/homebrew/opt/ffmpeg@4/bin/ffmpeg

~/Downloads 
➜ ffmpeg        
ffmpeg version 4.4.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with Apple clang version 13.0.0 (clang-1300.0.29.3)
  configuration: --prefix='/opt/homebrew/Cellar/ffmpeg@4/4.4.1' --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags= --enable-avresample --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'

~/Downloads 
❯ bash UVtools.app/Contents/MacOS/UVtools.sh 
UVtools.app/Contents/MacOS/UVtools.sh: line 3:  1383 Killed: 9               ./UVtools
UVtools.app/Contents/MacOS/UVtools.sh: line 3: dotnet: command not found

~/Downloads 
❯ otool -L UVtools.app/Contents/MacOS/UVtools   
UVtools.app/Contents/MacOS/UVtools:
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 904.4.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.60.1)
```

Let me know if there's some debugging/tracing I can do to track down the issue. I'll also see if I can get a local build going

sn4k3, 2022-03-13:
@bgyarfas the netframework is all the other files you see in UVtools directory, but as @rfield19 told, it is unable to run without install and run via dotnet, which can mean a missing dependency or missing permition, can you guys just chmod all files within uvtools and give them +x, if it fails then try give full permitions of 777?

rfield19, 2022-03-13:
> @bgyarfas the netframework is all the other files you see in UVtools directory, but as @rfield19 told, it is unable to run without install and run via dotnet, which can mean a missing dependency or [...]

Yep, tried that (set everything to 777 recursively) earlier to see if it would help! (it didn't...)
Same behaviour unfortunately - the uvtools binary dies immediately on launch.
[image]

sn4k3, 2022-03-14:
Maybe it should be compiled on arm?
Can you try to build it there?

- Clone this repo
- Open a terminal on UVtools.WPF folder
- Then run:
```bash
dotnet restore
dotnet publish $project -o "publish" -c Release -r osx-arm64 -p:PublishReadyToRun=true --self-contained
```
- After you need to copy libcvextern.dylib from my package to that folder
- Run ./UVtools

Does it works?

rfield19, 2022-03-14:
> ```shell
> dotnet publish $project -o "publish" -c Release -r osx-arm64 -p:PublishReadyToRun=true --self-contained
> ```

Blimey, compiled it - and running ./UVtools works...
But there's more!

Extrapolating from that, I then copied _all_ the files from the publish dir into the UVtools.app/Contents/MacOS dir - and launching it from the app icon then works as well 👍

I then tried (on a fresh download of the release) only copying the UVtools binary into the MacOS dir and that wasn't launchable from the icon, but was still runnable from the sh method.  

There must therefore be something in addition to the uvtools binary that it needs - and also must be necessary to compile the whole thing on an m1 machine I guess.

Do you want me to send over the working arm64 UVtools 3.0 binary?  I guess I can just attach it to this thread?

rfield19, 2022-03-14:
Link to new binary
https://we.tl/t-FnMTe3vo6U

rfield19, 2022-03-14:
... and from some further tests, the files that have changed in size on the new build compared to the release 3.0.0 are:

uvtools bin (of course)
UVtools.deps.json
UVtools.dll
UVtools.pdb
UVtools.runtimeconfig.json

sn4k3, 2022-03-14:
> There must therefore be something in addition to the uvtools binary that it needs - and also must be necessary to compile the whole thing on an m1 machine I guess.

Well that's not good news to me...

> UVtools.deps.json
> UVtools.pdb
> UVtools.runtimeconfig.json

Those files are not important

rfield19, 2022-03-14:
> > There must therefore be something in addition to the uvtools binary that it needs - and also must be necessary to compile the whole thing on an m1 machine I guess.
> 
> Well that's not good news to me... I'm gona ask on dotnet git
> 
> > UVtools.deps.json
> > UVtools.pdb
> > UVtools.runtimeconfig.json
> 
> Those files are not important, but the others are
> 
> I've updated the release build with yours :) How does it fell compiling such project and making work for mac users? ;)

Neat - I hope it works ok for others. 
@bgyarfas, maybe give it a go and see if it works for you also :)

It was extremely straightforward to compile it with your instructions though, so worst case m1 owners might have to compile their own version until/if you get an m1 machine.  

I'm not sure about what the other files are that it needs though.
I'll do a proper diff on the directory and see if there's anything else that's changed.

sn4k3, 2022-03-14:
I have asked here: https://github.com/dotnet/runtime/discussions/66574
If sign is required even your build will not work for others :/ 

> until/if you get an m1 machine.
I'm up for offers 😂

bgyarfas, 2022-03-14:
> Maybe it should be compiled on arm? Can you try to build it there?
>  ...
> Does it works?

Building from source worked for me as well. I have an Apple developer account so if I can find some time today I'll attempt to code-sign a version to see if @rfield19 can it. I suspect I'll need to build libcvextern as well. 

@sn4k3 I haven't looked, but I'm assuming there's tooling somewhere in the repo (or through `dotnet`) that ultimately creates the `.app` bundle?

As a completely temporary work around the x64 version works fine on the M1 (Rosetta 2). So for users really wanting 3.0.0 they just use [UVtools_osx-x64_v3.0.0.zip](https://github.com/sn4k3/UVtools/releases/download/v3.0.0/UVtools_osx-x64_v3.0.0.zip) for now.

sn4k3, 2022-03-14:
> I suspect I'll need to build libcvextern as well.

No, it's already compiled, you just need to codesign every binary, there are scripts for that

>  I haven't looked, but I'm assuming there's tooling somewhere in the repo (or through dotnet) that ultimately creates the .app bundle?

I have, but in a windows and powershell env. But creating a .app is very easy, since it's a folder with contents

Whitespace, 2022-03-18:
I'm on an M1 Pro running MacOS Monterey 12.2.1 trying to compile 3.1.0.

I think what y'all were proposing is the following.  LMK if there's something else to run.

```
# dependencies
brew install dotnet # 6.0.103
brew install cmake ffmpeg@4

# UVtools repo
git clone https://github.com/sn4k3/UVtools.git
cd UVtools

# build UVtools
cd UVtools.WPF
dotnet restore
dotnet publish -o "publish" -c Release -r osx-arm64 -p:PublishReadyToRun=true --self-contained
cp ../UVtools.Platforms/osx-arm64/libcvextern.dylib .

# run UVtools
./UVtools.sh
  zsh: permission denied: ./UVtools.sh

# max permissions on everything
chmod -R 777 ..
./UVtools.sh
  Could not execute because the specified command or file was not found.
  Possible reasons for this include:
    * You misspelled a built-in dotnet command.
    * You intended to execute a .NET program, but dotnet-UVtools.dll does not exist.
    * You intended to run a global tool, but a dotnet-prefixed executable with this name could not be found on the PATH.
```

rfield19, 2022-03-18:
@Whitespace just run `./UVtools` from the publish dir, rather than the sh script and it should work fine.
All the sh script does is cd to a dir and run ./UVtools anyway.

rfield19, 2022-03-18:
Incidentally, this is my workflow - as this copies the new files back into the UVtools app, and lets you run it normally from the regular uvtools icon again.

```# UVtools repo
git clone https://github.com/sn4k3/UVtools.git
cd UVtools

# build UVtools
cd UVtools.WPF
dotnet restore
dotnet publish -o "publish" -c Release -r osx-arm64 -p:PublishReadyToRun=true --self-contained

#reassemble app!
_cd publish_
_cp ~/Applications/UVtools.app/Content/MacOS/libcvextern.dylib ._
_yes | cp -rf * ~/Applications/UVtools.app/Contents/MacOS_
```

sn4k3, 2022-03-19:
@rfield19 i have created a humble compilation, publish and packing script with all the dependencies in place, produces the same as my release script: https://github.com/sn4k3/UVtools/blob/master/build/CreateRelease.sh

```sh
./CreateRelease.sh osx-x64
```

rfield19, 2022-03-19:
Unfortunately not having much luck with that script!
It comes up with an invalid grep parameter error, for the 
`version="$(grep -o '<Version>\K(\d\.\d\.\d)(?=<\/Version>)' $coreDir/UVtools.Core.csproj)"` 
line

sn4k3, 2022-03-19:
Ok maybe mac don't support that grep thing, let me test on my VM

sn4k3, 2022-03-19:
It seens mac system grep no longer support  -P
replace that line by this and report back if works: 

`version="$(perl -nle'print $& while m{<Version>\K(\d\.\d\.\d)(?=<\/Version>)}g' "$coreDir/UVtools.Core.csproj")"`

rfield19, 2022-03-19:
All went well until last bit of the script.

```
  UVtools.WPF -> /Users/rupert/Documents/Projects/UVtools/UVtools.WPF/bin/Release/net6.0/osx-arm64/UVtools.dll
  UVtools.WPF -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/
3. Copying dependencies
**xargs: illegal option -- i
usage: xargs [-0opt] [-E eofstr] [-I replstr [-R replacements] [-S replsize]]
             [-J replstr] [-L number] [-n number [-x]] [-P maxprocs]
             [-s size] [utility [argument ...]]**
4. Cleaning up
5. Setting Permissions
/Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/UVtools.sh
7. Skipping Zip
8. Completed
```

sn4k3, 2022-03-19:
Ok another thing mac dont support: `find "$runtimePlatformDir" -type f | grep -i lib | xargs -i cp -fv {} "$publishRuntimeDir/"`
Let me find an alternative

rfield19, 2022-03-19:
Found the xargs one!  

 -i[replace-str], --replace[=replace-str]
              This option is a synonym for -Ireplace-str if replace-str
              is specified.  If the replace-str argument is missing, the
              effect is the same as -I{}.  **This option is deprecated;
              use -I instead.**

rfield19, 2022-03-19:
Another xargs issue... 
Perhaps it needs the cp -fv command on that line to be in quotes or something?

xargs: illegal option -- f
usage: xargs [-0opt] [-E eofstr] [-I replstr [-R replacements] [-S replsize]]
             [-J replstr] [-L number] [-n number [-x]] [-P maxprocs]
             [-s size] [utility [argument ...]]

sn4k3, 2022-03-19:
Humm, lets drop that line and replace by: `cp -afv "$runtimePlatformDir/." "$publishRuntimeDir/"`

rfield19, 2022-03-19:
OK, it's running, but it's not executing step 6 - so it's creating the bin file, but not creating the app bundle.

ie `if [[ $runtime = osx-* ]]; then`...

3. Copying dependencies
/Users/rupert/Documents/Projects/UVtools/UVtools.Platforms/osx-arm64/. -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/.
/Users/rupert/Documents/Projects/UVtools/UVtools.Platforms/osx-arm64/./libcvextern.dylib -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/./libcvextern.dylib
/Users/rupert/Documents/Projects/UVtools/UVtools.Platforms/osx-arm64/./UVtools.entitlements -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/./UVtools.entitlements
/Users/rupert/Documents/Projects/UVtools/UVtools.Platforms/osx-arm64/./Info.plist -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/./Info.plist
4. Cleaning up
5. Setting Permissions
/Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/UVtools.sh
7. Skipping Zip
8. Completed

sn4k3, 2022-03-19:
Step 6 is zip, not required for you. But if you want to zip you need to pass -z flag `createRelease.sh -z osx-arm64`
To bundle into a mac app you need to use the -b flag `createRelease.sh -b osx-arm64`

rfield19, 2022-03-19:
Yes, that's what I meant - I did run it as 
`createrelease.sh -b osx-arm64` and it still skipped step 6 (the whole app bundle thing).

sn4k3, 2022-03-19:
if you using git: 

```bash
git pull
git reset --hard origin/master
```

rfield19, 2022-03-19:
Yep, I'm ok with git, just programming I suck at!

updated fork is still skipping step 6 after the `-b osx-arm64` argument.

```
  UVtools.WPF -> /Users/rupert/Documents/Projects/UVtools/UVtools.WPF/bin/Release/net6.0/osx-arm64/UVtools.dll
  UVtools.WPF -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/
3. Copying dependencies
/Users/rupert/Documents/Projects/UVtools/UVtools.Platforms/osx-arm64/. -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/.
/Users/rupert/Documents/Projects/UVtools/UVtools.Platforms/osx-arm64/./libcvextern.dylib -> /Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/./libcvextern.dylib
4. Cleaning up
5. Setting Permissions
/Users/rupert/Documents/Projects/UVtools/publish/UVtools_osx-arm64_v3.1.0/UVtools.sh
7. Skipping Zip
8. Completed
```

sn4k3, 2022-03-20:
Maybe another macOS difference...

Instead of `if [[ $runtime = osx-* ]]; then` try `if [[ $runtime == osx-* ]]; then`

rfield19, 2022-03-20:
It now starts step 6 - getting there!

error on this one next
"sed: -I or -i may not be used with stdin"

`sed -i "s/#VERSION/$version/g" "$osxApp/Contents/Info.plist"`

rfield19, 2022-03-20:
Someone suggested adding a zero length extension at the start of the sed line, and that works great
`sed -i '' "s/#VERSION/$version/g" "$osxApp/Contents/Info.plist"`

Script now runs fully, and creates a (working) regular app with icon in the publish dir.  Nice!

sn4k3, 2022-03-20:
Nice, i have updated the script, you may fetch to get everything in place (there are others fixes regarding ==)
Thanks for testing

rfield19, 2022-03-20:
No worries - happy to help!

sn4k3, 2022-03-20:
Now to fix the public releases for macOS i would need an developer account from Apple.
This will impact x64 too in future and not able to execute too

rfield19, 2022-03-20:
Don't know if this helps - you can apply for a free developer account if you're a non-profit 
https://developer.apple.com/support/membership-fee-waiver/

sn4k3, 2022-03-20:
> nonprofit organization, accredited educational institution

You need an registed organization, UVtools is just an app...

Also Portugal is not included

> Eligible regions: Australia, Brazil, Canada, China mainland, France, Germany, Israel, Italy, Japan, Mexico, South Korea, United Kingdom, United States.

illusio26, 2022-06-05:
I'm running into the same issue. I followed the install instructions on the main page, but no matter what I try, I get the "this application cannot be opened" prompt.

sn4k3, 2022-06-05:
@illusio26 There is a way if you compile it on your machine (readd all this discussion)
Or just run the x64 as that will still work

illusio26, 2022-06-05:
Thanks. I just ran with the x64 for now and that worked just fine.

ok1ok1, 2022-07-17:
> Nice, i have updated the script, you may fetch to get everything in place (there are others fixes regarding ==) Thanks for testing

Where is this script?

sn4k3, 2022-07-17:
> Where is this script?

You need to clone the repo, and it's located here: https://github.com/sn4k3/UVtools/tree/master/build 
Script: [createRelease.sh](https://github.com/sn4k3/UVtools/blob/master/build/createRelease.sh)

`./createRelease.sh -b osx-arm64`

[84 more replies]

closed 2023-04-23 by sn4k3