sn4k3/UVtools #432: [BUG] Automatic upgrade from version 2.29.0 to 3.0.0 does not work on Windows

tobecwb, 2022-03-12:
## System
 - OS: Windows 10 x64

```
## Describe the bug
I was using UVtools v2.29.0 today and I see a info on application that a new version was available (v3.0.0).
I answer yes to auto-update the application, the new version was downloaded and started automatically.

After finishing the update, UVTools no longer works.
Every time I click on the app it closes automatically.

Looking at the error log, I see the following information: _Could not load file or assembly 'Newtonsoft.Json, Version=12.0.0.0, Culture=neutral, PublicKeyToken=30ad4fe6b2a6aeed_'

The solution was to remove UVTools (via add/remove programs in the control panel), and install it again.

## To Reproduce
Steps to reproduce the behavior:

1. Start UVTools v2.29.0
2. Wait for update notice (version 3.0.0)
3. Complete the automatic update process
4. Start the UVTools

## Expected behavior
Launch UVTools in its latest version

[errors.log](https://github.com/sn4k3/UVtools/files/8238673/errors.log)

sn4k3, 2022-03-13:
Major versions require uninstall the prior and reinstall since it use a new framework version.
MSI handles upgrades very poorly

xPew, 2022-03-16:
@sn4k3 I had the same issue, and reinstalled UVtools automatically, to make sure the update didnt get messed up than headed here and saw its a legit issue. Maybe you should add a reminder for Windows user when the update button is clicked, to make sure its obvious if its a known limitation of the current msi.
Anyway, its a really nice update, honestly since I use UVtools, my resin prints are on the next lvl! Cant imagine where you update from here :)

sn4k3, 2022-03-16:
The issue is caused by a dependency that is not longer part of UVtools but part of other dependency, then files didn't match but MSI don't compare files to update them, which is bad. 
This also happen in other releases with other users but not all...
I will try to msi uninstall previous to install the new one

Hope that v3.1 fix this even from v2.9 to v3.1.
Also v3.1 will bring something new

> Anyway, its a really nice update, honestly since I use UVtools, my resin prints are on the next lvl! Cant imagine where you update from here :)

Well that's the objective, bring better prints and more safe prints to the users. Glad that is usefull for someone :)

closed 2022-03-18 by sn4k3