sn4k3/UVtools #439: 3.2.0 doesn't run on MacOS (x64)

jgrr, 2022-03-29:
Running from the commandline, I got the error: "The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10826 "kLSNoLaunchPermissionErr: User doesn't have permission to launch the app (managed networks)" UserInfo={_LSFunction=_LSLaunchWithRunningboard, _LSLine=2488, NSUnderlyingError=0x7fcce7a0ce50 {Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x7fcce7a07650 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111" UserInfo={NSLocalizedDescription=Launchd job spawn failed with error: 111}}}}}"

I was able to fix it by doing chmod +x on UVTools.app/Contents/MacOS/UVTools. Not sure if there is anything else that needs permissions fixed.

sn4k3, 2022-03-29:
This will become harder and harder without codesign the application. Apple is enforcing that. 
All i can do is try to raize funds to buy apple developer and also an mac to fix this without user interaction. 
Also #431

sn4k3, 2022-04-05:
it seens chmod solve on x64: https://github.com/sn4k3/UVtools/discussions/447

closed 2022-04-05 by sn4k3

r-a-i, 2022-12-03:
Hi, I just experienced this when migrating to a new M1/Arm Mac and trying to run UVTool 3.9.
The previous version 3.1 (as per #447) does work without having to chmod.
I did not try any intermediate versions after 3.1.