sn4k3/UVtools #441: [BUG] LiftHeight property is always 6 for Anycubic Photon Ultra

manolitto, 2022-03-30:
## System
 - OS: Mac OS X Moneterey
 - Slicer: PrusaSlicer 2.4.1
 - Printer: Anycubic Photon Ultra
 - UVtools version: 3.2.0 X64

## Describe the bug
Whatever value I set as LiftHeight in Prusa Printer profile, the resulting SL1/DLP files always have a "LiftHeight" property of 6

Here is my Notes section from my custom printer profile in PrusaSlicer (see also Screenshot):

```
Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.
PRINTER_VENDOR_PRUSA3D
PRINTER_MODEL_SL1
PRINTER_VENDOR_ANYCUBIC
PRINTER_MODEL_PHOTON_ULTRA
FILEFORMAT_DLP

START_CUSTOM_VALUES
BottomLiftHeight_5
BottomLiftSpeed_120
LiftHeight_5
LiftSpeed_120
BottomLiftHeight2_5
BottomLiftSpeed2_120
LiftHeight2_5
LiftSpeed2_120
BottomRetractSpeed_120
RetractSpeed_120
BottomLightPWM_255
LightPWM_255
WaitTimeBeforeCure_1
BottomWaitTimeBeforeCure_1
END_CUSTOM_VALUES
```

## To Reproduce
Steps to reproduce the behavior:
1. Import "AnyCubic Photon Ultra" profile in PrusaSlicer
2. Use a "LiftHeight_5" value in printer profile notes in PrusaSlicer (or any other value than 6)
3. Export SL1 file
4. Open file in UVtools
5. See "LiftHeight" property value

## Expected behavior
Expected "LiftHeight" property value in UVtools is 5

## Screenshots
[image]
[image]

sn4k3, 2022-03-30:
I see the problem, it's also present on bottom lift. Will fix on next patch

sn4k3, 2022-04-02:
Fixed

closed 2022-04-02 by sn4k3

manolitto, 2022-04-02:
Works great! Thanks for the quick fix and your super-helpful tool! Saves my life every day. :-)