sn4k3/UVtools #444: [FEATURE REQUEST]

hugoalves6, 2022-04-02:
Hello

I started to use UVTools, and loved it right away, most print's i've gotten to fail due to some user errors, got fixed with this tool, and I love even more that I can slice the object in any of the slicers I use most, and it just auto convert them.

Now I was wandering if it is possible, to implement a feature that we (users) could check in the automation options, that after it converts a file, to simply remove the old one.

EX:
on prusa slicer, i save a SL1 file, UVTools convert to CTB, after converting it, remove the old SL1 file.

Because after the convertion on UVTools any edition it's already made on the final file (in my case CTB files).

sn4k3, 2022-04-02:
Sure, it's a good to have

hugoalves6, 2022-04-03:
Thanks, it's just to simplify the file management, and to avoid scrapyard files xD

closed 2022-04-06 by sn4k3