sn4k3/UVtools #446: Getting "Repair layers and errors" popup error window when loading *.cbt file Windows 10

orosalsero1, 2022-04-04:
## System
 - OS: Windows 10 x64
 - Slicer: Chitubox v.1.9.2
 - Printer: QIDI S-box
 - UVtools version: 3.2.1

```
UVtools v3.2.1
Operative system: Microsoft Windows 10.0.19043 X64
Processor: Intel(R) Core(TM) i5-6400 CPU @ 2.70GHz
Processor cores: 4
Memory RAM: 7.15 / 11.90 GB
Runtime: win10-x64
Framework: .NET 6.0.3
AvaloniaUI: 0.10.13
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080
2: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1080    UA: 1920 x 1080

Path: C:\Program Files\UVtools_win-x64_v3.2.1\

General configuration for OpenCV 4.5.5 =====================================
  Version control:               4.5.5-237-gaa8b5d2f70

  Extra modules:
    Location (extra):            D:/bb/cv_x64/build/build_x64/../opencv_contrib/modules
    Version control (extra):     4.5.5-62-g7cb6b4d5

  Platform:
    Timestamp:                   2022-01-05T15:01:30Z
    Host:                        Windows 10.0.22000 AMD64
    CMake:                       3.21.3
    CMake generator:             Visual Studio 16 2019
    CMake build tool:            C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
    MSVC:                        1929
    Configuration:               Debug Release MinSizeRel RelWithDebInfo

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3
      requested:                 SSE3

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe  (ver 19.29.30138.0)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MD /O2 /Ob2 /DNDEBUG 
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W4 /GR  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /EHa /wd4127 /wd4251 /wd4324 /wd4275 /wd4512 /wd4589 /MP  /MDd /Zi /Ob0 /Od /RTC1 
    C Compiler:                  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP   /MD /O2 /Ob2 /DNDEBUG 
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3  /D _CRT_SECURE_NO_DEPRECATE /D _CRT_NONSTDC_NO_DEPRECATE /D _SCL_SECURE_NO_WARNINGS /Gy /bigobj /Oi  /fp:precise     /MP /MDd /Zi /Ob0 /Od /RTC1 
    Linker flags (Release):      /machine:x64  /INCREMENTAL:NO 
    Linker flags (Debug):        /machine:x64  /debug /INCREMENTAL 
    ccache:                      NO
    Precompiled headers:         YES
    Extra dependencies:          wsock32 comctl32 gdi32 ole32 setupapi ws2_32 D:/bb/cv_x64/build/build_x64/install/lib/freetype.lib D:/bb/cv_x64/build/build_x64/install/lib/harfbuzz.lib D:/bb/cv_x64/build/build_x64/install/lib/libhdf5.lib vtkRenderingOpenGL2 vtkCommonCore vtksys vtkCommonDataModel vtkCommonMath vtkCommonMisc vtkCommonSystem vtkCommonTransforms vtkCommonExecutionModel vtkRenderingCore vtkCommonColor vtkCommonComputationalGeometry vtkFiltersCore vtkFiltersGeneral vtkFiltersGeometry vtkFiltersSources vtkglew vtkInteractionStyle vtkFiltersExtraction vtkFiltersStatistics vtkImagingFourier vtkImagingCore vtkRenderingLOD vtkFiltersModeling vtkIOPLY vtkIOCore vtkdoubleconversion vtklz4 vtklzma vtkzlib vtkFiltersTexture vtkRenderingFreeType vtkfreetype vtkIOExport vtkIOImage vtkDICOMParser vtkmetaio vtkjpeg vtkpng vtktiff vtkIOXML vtkIOXMLParser vtkexpat vtkRenderingContext2D vtkRenderingGL2PSOpenGL2 vtkgl2ps vtkIOGeometry vtkIOLegacy
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libpng libtiff libopenjp2 IlmImf zlib quirc

  OpenCV modules:
    To be built:                 alphamat aruco barcode bgsegm bioinspired calib3d ccalib core datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hdf hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking ts video videoio videostab viz wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    python_bindings_generator python_tests world
    Disabled by dependency:      -
    Unavailable:                 cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev cvv java julia matlab ovis python2 python3 sfm
    Applications:                perf_tests
    Documentation:               NO
    Non-free algorithms:         NO

[... 237 lines cut ...]

 ---> (Inner Exception #6) System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.Operations.OperationRepairLayers.<>c__DisplayClass69_2.<ExecuteInternally>b__7(IGrouping`2 group) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationRepairLayers.cs:line 322
   at System.Threading.Tasks.Parallel.<>c__DisplayClass44_0`2.<PartitionerForEachWorker>b__1(IEnumerator& partitionState, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass44_0`2.<PartitionerForEachWorker>b__1(IEnumerator& partitionState, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica`1.ExecuteAction(Boolean& yieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()<---

 ---> (Inner Exception #7) System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.Operations.OperationRepairLayers.<>c__DisplayClass69_2.<ExecuteInternally>b__7(IGrouping`2 group) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationRepairLayers.cs:line 322
   at System.Threading.Tasks.Parallel.<>c__DisplayClass44_0`2.<PartitionerForEachWorker>b__1(IEnumerator& partitionState, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass44_0`2.<PartitionerForEachWorker>b__1(IEnumerator& partitionState, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica`1.ExecuteAction(Boolean& yieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()<---

## Expected behavior
I close the error window and I choose repair file than nothing the timer starts on file repair but the bar is always at 68%

## A clear and concise description of what you expected to happen.
I expect the bar to move as the file is being fixed but it just stays at 68% for minutes on end.

## Screenshots
[image]

sn4k3, 2022-04-04:
I think your **Steps To Reproduce** the error are incomplete. 
I can open the file correctly, i also search issues and repair issues without any trouble.
Can you update the steps to the exact steps?

sn4k3, 2022-04-06:
Fixed

closed 2022-04-06 by sn4k3