sn4k3/UVtools #448: [BUG] Anycubic Mono X 6K images are cut and shift on X-Axis every few layers.

Susanoo1337, 2022-04-04:
## System
 - Slicer: Prusa
 - Printer: Anycubic Mono X 6K
 - UVtools version: 3.2.1

Bug & Steps to reproduce:
1. STL Sliced in Prusa Slicer, saved as .sl1, slice looked ok in preview.
2. Opened in UVTools, Auto converted to .pwmb, slice looked on in preview.
3. Checked preview in Photon Workshop, looked ok.
4. During printing, on certain layers, part of the image is shifted to the side. (X axis)

## Screenshots
Layer 16(?)* printer:
[image]
Layer 16 UVTools:
[image]
Layer 39 printer(The cut off and shifted part was out of frame, all the way to the left of the screen):
[image]
Layer 39 UVTools:
[image]

Layer 27(?)* picture was unclear, but same issue.

(*not 100% sure on the layer numbers 16 and 27, can be +/-1, only what the printer showed on the display while I was watching the UV light and filming at the same time, layer 39 however is 100% defective, there were probably more layers but I didn't film/watch the entire print to the end)

## Files
pwmb file in question
[test_copy.zip](https://github.com/sn4k3/UVtools/files/8410068/test_copy.zip)

## Additional context
This issue was consistent over the course of the past 3 days, across 3 versions of UVTools. I received this printer very recently, having been FDM printing for a long time I'm very familiar with PrusaSlicer and opted to use it in conjunction with UVTools over PhotonWorkshop or ShituBox. While calibrating exposure for my resin I thought it's a layer adhesion issue, it wasn't until today I decided to look at what the display was actually showing. Especially after the same object sliced with PhotonWorkshop produced a normal print. Almost lost my mind. Not complaining, I'd love to continue using Prusa+UVTools, but this is a serious issue as it automatically fails prints.

sn4k3, 2022-04-04:
Interesting... Did you change any settings other than defaults?

Susanoo1337, 2022-04-05:
I didn't really touch any of the settings afaik, installed, imported profiles to PrusaSlicer and started slicing. I've changed print settings and tried some built in calibration prints but so far everything failed, I presume due to this issue.
I use PNG compression, I might try the others and see if the issue persists.

Susanoo1337, 2022-04-05:
Preeliminary testing indicates that only files sliced in Prusa as sl1 and then converted to pmwb suffer from this issue. I can slice in PhotonWorkshop and then edit print settings and pixels in UVTools seemingly without issue. 
I couldn't test too much today since I ran out of gloves and couldn't run prints, so I've watched some of the files directly on the LCD screen. But that's really boring and sometimes the issue doesn't show up early.

If I run across the issue again with files sliced in PhotonWorkshop I'll try your 1. suggestion.
Sadly I don't know anyone with another Mono X 6K. I thought it might be a mainboard issue or a screen issue, if I can't rule that out I'll try asking around the 3D printing discord.

sn4k3, 2022-04-05:
`...indicates that only  files sliced in Prusa as sl1`, in the end of the day, encoding sl1 to pwmb is the same as encoding pwmb to pwmb (aka Save)

`...edit print settings` dont raise a full encode. But editing pixels will do. You can also try re-convert the file from .pwmb to .pwmb, selecting v516.

`...I've watched some of the files directly on the LCD screen` Thats the easiest and fastest way of debuging, dry prints.

`Sadly I don't know anyone with another Mono X 6K.` I can ask on group and redirect to here

IMO something is happening on printer side, because line is there, if UVtools show it correctly and if you open on thier slicer is also present then may be something printer side

[image]

I would also print same file multiple times and log which layers have the trouble, if different prints show different layers then is printer side

white-stormy, 2022-04-05:
this is a known bug for this printer. firmware update fixes it. as of right now you might have to do their after sales service ticket because they are still holding back on releasing their latest for one of the two mother boards they used.

sn4k3, 2022-04-05:
Also the discussion here: https://www.facebook.com/groups/uvtools/posts/656614615448045/

antwal, 2022-04-06:
I got the mono x 6k today, I'm trying it but this problem does not occur, I will do some other tests tomorrow about it

sn4k3, 2022-04-10:
Closed - Printer firmware issue

closed 2022-04-10 by sn4k3