sn4k3/UVtools #449: [BUG] Incorrect calculation material cost after fix issues

antwal, 2022-04-05:
## System
 - OS: macOS 10.15.7
 - Slicer: PrusaSlicer 2.4.1
 - Printer: ELEGOO Saturn
 - UVtools version: 3.2.1

```
UVtools v3.2.1
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Feb 15 21:39:11 PST 2022; root:xnu-6153.141.59~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz
Processor cores: 16
Memory RAM: 34.24 / 64.12 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.3
AvaloniaUI: 0.10.13
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 200% (Primary)
    WA: 2560 x 1417    UA: 1280 x 708
2: 2560 x 1440 @ 200%
    WA: 2560 x 1417    UA: 1280 x 708
3: 2560 x 1440 @ 200%
    WA: 2560 x 1417    UA: 1280 x 708

Path: /Applications/UVtools.app/Contents/MacOS/

General configuration for OpenCV 4.5.4 =====================================
  Version control:               4.5.4-235-g27567928d7

  Extra modules:
    Location (extra):            /Users/tiago/Desktop/emgucv/opencv_contrib/modules
    Version control (extra):     4.5.4-61-gc33e5b60

  Platform:
    Timestamp:                   2021-11-01T19:19:02Z
    Host:                        Darwin 18.7.0 x86_64
    CMake:                       3.21.4
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               Release

  CPU/HW features:
    Baseline:                    SSE SSE2 SSE3 SSSE3 SSE4_1
      requested:                 DETECT
    Dispatched code generation:  SSE4_2 FP16 AVX AVX2 AVX512_SKX
      requested:                 SSE4_1 SSE4_2 AVX FP16 AVX2 AVX512_SKX
      SSE4_2 (1 files):          + POPCNT SSE4_2
      FP16 (0 files):            + POPCNT SSE4_2 FP16 AVX
      AVX (4 files):             + POPCNT SSE4_2 AVX
      AVX2 (30 files):           + POPCNT SSE4_2 FP16 FMA3 AVX AVX2
      AVX512_SKX (5 files):      + POPCNT SSE4_2 FP16 FMA3 AVX AVX2 AVX_512F AVX512_COMMON AVX512_SKX

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /Library/Developer/CommandLineTools/usr/bin/c++  (ver 10.0.1.10010046)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /Library/Developer/CommandLineTools/usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winconsistent-missing-override -Wno-delete-non-virtual-dtor -Wno-unnamed-type-template-args -Wno-comment -fdiagnostics-show-option -Wno-long-long -Qunused-arguments -Wno-semicolon-before-method-body -ffunction-sections -fdata-sections  -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,-dead_strip  
    Linker flags (Debug):        -Wl,-dead_strip  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          -framework OpenCL -framework Cocoa -framework AppKit /Users/tiago/Desktop/emgucv/platforms/macos/build/install/lib/libhdf5.a Iconv::Iconv
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libpng libtiff libopenjp2 IlmImf zlib quirc

  OpenCV modules:
[... 148 lines cut ...]
089: System.Reflection, Version=6.0.0.0
090: System.Reflection.Extensions, Version=6.0.0.0
091: System.Net.Sockets, Version=6.0.0.0
092: Microsoft.Win32.Primitives, Version=6.0.0.0
093: System.Threading.ThreadPool, Version=6.0.0.0
094: System.Net.NameResolution, Version=6.0.0.0
095: System.IO.Compression.ZipFile, Version=6.0.0.0
096: System.Threading.Tasks.Parallel, Version=6.0.0.0
097: System.Security.Cryptography.Primitives, Version=6.0.0.0
098: System.Security.Cryptography.Encoding, Version=6.0.0.0
099: System.Formats.Asn1, Version=6.0.0.0
100: System.Runtime.Numerics, Version=6.0.0.0
101: System.IO.FileSystem.DriveInfo, Version=6.0.0.0
102: K4os.Compression.LZ4, Version=1.2.16.0
103: Markdown.Avalonia, Version=0.10.10.0
104: Markdown.Avalonia.SyntaxHigh, Version=0.10.10.0
105: System.Diagnostics.StackTrace, Version=6.0.0.0
106: ColorTextBlock.Avalonia, Version=0.10.10.0
107: AvaloniaEdit, Version=0.10.0.0
108: System.Diagnostics.Process, Version=6.0.0.0
109: System.IO.Pipes, Version=6.0.0.0
```

## Describe the bug
Open SL1 file, click on issues, click on repair issues, the material used is update but the price no

sn4k3, 2022-04-05:
MaterialMl is calculated independently and do not require external user variables, but not the price. 
The price information comes from SL1 file information and is one of the few formats who output resin price information. This information not even show on most of other formats. 
The only way to calculate price is to know used resin price, which UVtools can not know. While SL1 have such information others don't and can't be abstracted. 

However i can do math calculation to update price on the few supported files, if that's important

antwal, 2022-04-06:
I understand, I purposely use prusaslicer because it is one of the few that calculates the price; on some files corrected with uvtools initially they were 98ml after the fix of the problems they had become 170ml, but the price was not recalculated.

I could do the calculation manually but since the price has been implemented it would be convenient to have it recalculated.

sn4k3, 2022-04-06:
Implemented

closed 2022-04-06 by sn4k3

antwal, 2022-07-27:
the problem has come back, I will have to create an app that analyzes the file in order to have the precise calculation of the costs; the material, as always, is calculated correctly

sn4k3, 2022-07-27:
The calculation have not changed since v3.2.2.
I can't observe the problem, please attach a file that does.
Also provide more information, price before layer fix and after fix, what auto fixes do you run and any other task you done.

antwal, 2022-07-28:
> The calculation have not changed since v3.2.2. I can't observe the problem, please attach a file that does and the workflow. Also provide more information, price before layer fix and after fix, what auto fixes do you run and any other task you done.

I am currently using version 3.3.2 which works best even though it has several bugs and is not calculated correctly