sn4k3/UVtools #452: [BUG] Get white pixel where should not be placed while deleting island with issues repair

francescopela, 2022-04-07:
- Slicer: Photon workshop 2.1.29
 - Printer: Photon Mono SE
 - UVtools version:  3.2.2

Operative system: Microsoft Windows 10.0.19043 X64
Processor: Intel(R) Core(TM) i5-7300HQ CPU @ 2.50GHz
Processor cores: 4
Memory RAM: 6.50 / 7.87 GB
Runtime: win10-x64
Framework: .NET 6.0.3
AvaloniaUI: 0.10.13
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 125% (Primary) (On this)
    WA: 1920 x 1030    UA: 1536 x 824

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

```

## Describe the bug

When I delete Island of a sliced files, I get extra white pixel between existing white area (like, between supports or attaching a support to a nearby wall of the model that should not be there

## To Reproduce
Steps to reproduce the behavior:
repair a sliced file island with the repair tool

## Expected behavior
<!--
Only remove island, not extra white pixels where island are not involved !-->

## Screenshots
Original: before the repair process
Result: after repair process: as you can see they appear different area with white pixel between them, joining with "string" different support or attaching nearby pieces

[original](https://user-images.githubusercontent.com/103222312/162326685-e7ae1d83-2797-4c92-92cb-c88b31b67046.jpg)
![result](https://user-images.githubusercontent.com/103222312/162326785-65f35f7d-64c1-451f-bab0-244938f802c5.jpg)

francescopela, 2022-04-07:
I understood where the problem was.
[image]

In the advanced options of the  repair issue menu,  I had to remove the gap closing interaction.

Maybe it was a new option, I don't remember it in my first "exploration" of the software.

closed 2022-04-07 by francescopela

sn4k3, 2022-04-07:
That's normal and it's how it repair islands/layers, by attaching near by masses (It dont look for islands only, but all masses)
It come from gap closing iterations on repair dialog or settings.

> the problem

It's a feature, not a problem. Attach nearby supports yields no problem but also give better stability.

> Maybe it was a new option  

It's there since the begining.