sn4k3/UVtools #453: [BUG]

rebeltaz, 2022-04-11:
OS: Ubuntu 18.04 LTS
Appimge: UVtools_linux-x64_v3.3.0.AppImage

I have set the appimage as executable. I already have all of the required libraries installed with the exception of two - libgeotiff5 and libopenexr24 - which ubuntu repositories report as unable to find. Trying to run the appimage anyway generates this error:

```
Unhandled exception. System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'cvextern' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libcvextern: cannot open shared object file: No such file or directory
   at Emgu.CV.CvInvoke.RedirectError(CvErrorCallback errorHandler, IntPtr userdata, IntPtr prevUserdata)
   at Emgu.CV.CvInvoke..cctor()
   --- End of inner exception stack trace ---
   at Emgu.CV.CvInvoke.Init()
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/App.axaml.cs:line 211
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run() in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 181
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.X11.X11PlatformThreading.CheckSignaled() in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 164
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.X11/X11PlatformThreading.cs:line 244
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/Program.cs:line 60
Aborted
```

sn4k3, 2022-04-11:
Without the dependencies you will not be able to run it.
Ubuntu 18 is no longer supported by openCV that requires ubuntu 20 at least.

You can try `sudo apt install libtiff5 libopenexr22` but i guess it will fail due older libs.
The solution will be to upgrade Ubuntu to the recent LTS or newer

rebeltaz, 2022-04-11:
Yeah.. no. Those are already installed and at those versions. Thanks anyway.

closed 2022-04-11 by rebeltaz