sn4k3/UVtools #457: [BUG] Mono X 6K Lift distance not saved correctly.

Skyber1967, 2022-04-12:
## System
 - OS: Windows 10 x64 v21H2
 - Slicer: Lychee
 - Printer: Anycubic Mono X 6K
 - UVtools version: 3.3.1

```

## Describe the bug
After slicing with Lychee I open the .pwmd file with UVtools to check for problems, I let UVtools made corrections and then save the .pwmd .  When I print the original export file the lift distance is ok (6 > 2) 8mm. But when printing the UVtools saved .pwmd the lift distance is only 2mm and like attended gives problems. 

## Files
<!--
I added it Zip the original export from Lychee and the saved version from UVTools
!-->
 
[Bureau.zip](https://github.com/sn4k3/UVtools/files/8473118/Bureau.zip)

sn4k3, 2022-04-12:
In fact your original file have lift of 2mm every layer. It looks like a slicer bug and not UVtools.
See:

[image]

[image]

If the original file prints ok, then it's some value that is skipping that layer value and only following the main table which in UVtools file is also 0 (Skip per layer settings)

I will keep dig in this, but can you export stl from lychee and slice on photon workshop, process on uvtools and try if the outcome is the same?

Skyber1967, 2022-04-13:
Hi, I already did an stl export and slicing with Photon, I attached the stl and the Photon sliced .pwmd  
Maybe it could help but what I also saw on my printer when printing the files is that with the working files I can see in settings the double values for lift distance and speed (2 > 6), with the UVtools file only the first value is visible and editable. Maybe something wrong in the metadata ?  
[Photon.zip](https://github.com/sn4k3/UVtools/files/8479154/Photon.zip)

sn4k3, 2022-04-13:
I only see one difference between files on a field i dont know the meaning. I try to change it to 1. Please try the file and report back:
[Coins avec plaque uvtools 33.zip](https://github.com/sn4k3/UVtools/files/8482149/Coins.avec.plaque.uvtools.33.zip)

Skyber1967, 2022-04-13:
Just tested the file you sended and ...  IT WORKS !!! :)
The double parameters are now visible and the printer reacts normally (2 >6) 8mm.
This field seems to be important ;)
A guy from Lychee reacts also to my question and proposed to contact you, they had a similar problem with the pre-release of the Mono X 6K.

sn4k3, 2022-04-13:
Ok, good to know, i will patch accordingly. 
That was the only difference, if that didn't cut i was lost :P

sn4k3, 2022-04-14:
Fixed

closed 2022-04-14 by sn4k3