sn4k3/UVtools #459: [BUG] Files cannot be directly loaded by double click or "Open With" in Mac OS Finder

manolitto, 2022-04-13:
## System
 - OS: MacOS Monterey 12.3.1 
 - UVTools version: 3.3.1 X64

## Describe the bug
Slicer files (SL1, DLP, ...) cannot be loaded directly from Finder.
However, drag and drop works and Opening from UVTools works as expected.

## To Reproduce
Steps to reproduce the behaviour:
1. Locate an SL1 file in Finder
2. Right click file an chose "Open With" -> "UVTools.app"
    --> UVTools opens but file is not loaded
3. Drag and drop file from Finder to UVTools window
    --> file gets loaded without issues

## Expected behaviour
Upon double click or "Open With" that file should be loaded in a (newly opened?) UVTools instance.

sn4k3, 2022-04-13:
The double click on file to open is not a feature of UVtools. 
On windows is possible to register files to open with double click but i guess makes more sense to leave those to own slicer. Yet i can provide that in the future.
Regarding MacOS i have 0 experience as i never used, but i can give a look why open with is not working, maybe it's not redirecting args, i will take a look

sn4k3, 2026-09-19:
Can test last release v7 and report back if is fixed?

closed 2026-10-08 by sn4k3