sn4k3/UVtools #460: [FEATURE REQUEST] Horizontal limiting margin > 1000px

manolitto, 2022-04-13:
On very wide screens (mine is 5120x1440 px) a newly opened UVTools instance is way too big.
Unfortunately it is not possible to set a higher "Horizontal limiting margin" greater than 1000px.
Allowing a higher value would solve this issue for ultra-wide displays.

sn4k3, 2022-04-13:
Horizontal margin is to limit the fixed windows so they cant be sider than screen - margin.
But they scale acordingly, it should not expand more than required, if windows is 500px, it should only expand up to 500px.
Can you provide a screen shot?

manolitto, 2022-04-13:
Thought that this margin limiting feature also works for the main window, but from how I understand your answer, this limiting margin only applies to dialog windows?
Anyway this would be a nice feature if it could also limit the maximum size for the main application window. See screenshot to know what I mean. ;-)

But, of course, this is only a "nice to have" to make your awesome tool even more perfect.

[image]

sn4k3, 2022-04-13:
All windows that can be resizable don't have such limit property.
I can't tell you why does it spawns so large, you can opt to start maximized which is what most people use, but in normal state size is auto determined by OS and resolution, this come from framework not uvtools.
Main Window don't have any predefined size is auto. How sizing is being made don't depend on me.
But looks like correct, on my PC windows window is about 0.75x the screen in normal mode, your looks like same ratio or similar.

sn4k3, 2022-04-14:
Implemented a candy on settings

closed 2022-04-14 by sn4k3

manolitto, 2022-04-14:
Awesome. Thanks!