sn4k3/UVtools #467: UVTools not functional on Ubuntu Linux

frankvdh, 2022-04-25:
I did go through the build of libcvextern as described in https://opensourcelibs.com/lib/uvtools, but still get 
`frank@desktop:~/Downloads$ ./UVtools_linux-x64_v3.3.2.AppImage 
System.TypeInitializationException: The type initializer for 'Emgu.CV.CvInvoke' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'dl' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: libdl: cannot open shared object file: No such file or directory
   at Emgu.Util.Toolbox.Dlopen(String dllname, Int32 mode)
   at Emgu.Util.Toolbox.LoadLibrary(String dllname)
   at Emgu.CV.CvInvoke.LoadUnmanagedModules(String loadDirectory, String[] unmanagedModules)
   at Emgu.CV.CvInvoke.DefaultLoadUnmanagedModules(String[] modules)
   at Emgu.CV.CvInvoke..cctor()
   --- End of inner exception stack trace ---
   at Emgu.CV.CvInvoke.Init()
   at UVtools.WPF.App.OnFrameworkInitializationCompleted() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/App.axaml.cs:line 213
cannot open path of the current working directory: Permission denied
`
The GUI window says
`UVtools v3.3.2 X64  [Linux] 
Unable to run due one or more missing dependencies. 
Triggered by: libcvextern  (OpenCV)`

I'm wondering if libcvextern installation steps described apply if UVTools is installed in an AppImage? (and why libcvextern isn't included in the AppImage?)

Or maybe I installed libcvextern wrongly? I did the whole libcvextern build as described in https://opensourcelibs.com/lib/uvtools in a subdirectory of my user home directory. Does that need to be done by a superuser? When the build completed, apparently successfully, I tried running UVTools. When that failed, as per the instructions I copied libcvextern.so from ~/repos/emgucv/libs/runtimes/ubuntu-x64/native (not emgucv/libs/x64 which didn't exist) to the same directory as UVtools_linux-x64_v3.3.2.AppImage but that didn't help. libcvextern.so has rwxr-xr-x permissions.

I found another libcvextern.so in emgucv/Emgu.CV.Test/Emgu.CV.Test.Netcore/bin/Release/net6.0/ and (clutching at straws) I copied that to the same directory as the AppImage, but wasn't surprised when that still wouldn't run.

All in all, I suspect that there's something not quite right in the Ubuntu/Mint/Debian/Similars installation instructions, but maybe I've misunderstood something.

sn4k3, 2022-04-25:
⚠️ Missing crucial information: Your **OS** system and **version**

> https://opensourcelibs.com/lib/uvtools

That is not the official website, and then information can be outdated. Use https://github.com/sn4k3/UVtools instead.

> I'm wondering if libcvextern installation steps described apply if UVTools is installed in an AppImage (and why libcvextern isn't included in the AppImage?)

It does, but an AppImage is just like an zip which when you run it got extracted to a temporary folder and run from there. 
If you build libcvextern.so you need to repack it on AppImage, or just download the .zip uvtools package (Non appimage) and replace file there.
AppImage is just more convenient because you can run from single file instead of having a folder with contents.

> Or maybe I installed libcvextern wrongly?

The lib is not to install, but build and replace on UVtools directory/contents, but your problem most likely is dependencies.
Did you follow https://github.com/sn4k3/UVtools#ubuntumintdebiansimilars steps?

frankvdh, 2022-04-26:
Magic! It works. So sorry to trouble you.
Unfortunately the opensourcelibs link comes up first in a Google search for UVTools. Equally unfortunately, there's no way to give them feedback about their out-of-date documentation.

NB... libdc1394-22 & libopenexr24 have been superceded in Ubuntu.

Once again, many thanks.

Frank

closed 2022-04-26 by frankvdh