sn4k3/UVtools #468: [FEATURE REQUEST]

sjhan12, 2022-04-25:
Hi Tiago

I have one suggestion. 

In UVtoolsCmd.exe, please put a option to choose cbt version. 
 
For unencrypted ctb format, UVtoolsCmd.exe convert <input file name> cbt 3 <output file name> and for encrypted ctb format, UVtoolsCmd.exe  convert <input file name> cbt -1 <output file name>

That would make UVtoolsCmd.exe much easy to use. 

Thank you for the great program. 

Sangjo

Also look at: https://github.com/sn4k3/UVtools/discussions/categories/ideas

## Is your feature request related to a problem? Please describe
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

## Describe the solution you'd like
A clear and concise description of what you want to happen.

## Describe alternatives you've considered

## Screenshots

## Additional context

sn4k3, 2022-04-25:
That already exists.
Each command have it own flags, to know them just do: `UVtoolsCmd.exe thecommand /?`

[image]

So all you need is: `UVtoolsCmd.exe convert -v 4 yourfile.from Chitubox yourfile.ctb`
Since .ctb extensions exists on two distinct classes/formats you need to strict tell the class name, which you can consult here: https://github.com/sn4k3/UVtools/tree/master/UVtools.Core/FileFormats

For ctb encrypted: `UVtoolsCmd.exe convert yourfile.from CTBEncrypted yourfile.ctb`

sjhan12, 2022-04-25:
Great!

I didn't see it. Thank you for letting me know.

closed 2022-04-25 by sn4k3