sn4k3/UVtools #469: [BUG] "Wait time before cure" recommendation is factoring lift and retract time when it shouldn't and also is not inserting a blank first layer.

WestonHawk, 2022-05-04:
## System
 - OS: Windows 10 x64 v20H2
 - Slicer: Lychee
 - Printer: Elegoo Saturn
 - UVtools version: 3.41

```
UVtools v3.4.1 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: AMD Ryzen 9 5900X 12-Core Processor            
Processor cores: 24
Memory RAM: 10.19 / 31.92 GB
Runtime: win10-x64
Framework: .NET 6.0.4
AvaloniaUI: 0.10.13
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 3840 x 1600 @ 100% (Primary) (On this)
    WA: 3840 x 1560    UA: 3840 x 1560
2: 2560 x 1440 @ 100%
    WA: 2560 x 1440    UA: 2560 x 1440

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

```

## Problem 1:

When applying the "Wait time before cure: 20/2s" recommendation it is also taking into account the time to lift and retract, which is not necessary on my printer.

Real world example:

When I apply the "Wait time before cure: 20/2s" it sets BottomLightOffDelay to `25.33s` and LightOffDelay to `6.5s` when I expect it to set them to `20s` and `2s` respectively.

My bottom and normal lift distance is `5mm`. Bottom lift speed is `1.5mm/s` and normal lift speed is `2mm/s`. My bottom and normal retract speed are `2.5mm/s`. Total lift and retract time is then `5.33s` for bottom layers and `4.5s` for normal layers. That is exactly what it is adding to the delay time.

BottomLightOffDelay and LightOffDelay on my printer (Elegoo Saturn) are not timed until *after* the bed has lifted and retracted. There is no need to take the lift and retract time into account for light off delay.

## Problem 2:

I have "Create a first empty layer to bypass printer limitation" checked under the "Wait time before cure" suggestions settings, but applying the suggestion does not create an empty first layer. The total layer count does not increase so I can confirm it is not being placed elsewhere.

## Related feature request:

I would like the option to insert a blank layer anytime. This could be placed under the "Actions" menu alongside the clone layer option. I usually already have my delays set how I want when I export from Lychee and only need to insert a blank first layer. Duplicating the first layer then having to blank it out through pixel arithmetic is cumbersome with all the options I have to adjust and deselect in there.

tslater2006, 2022-05-04:
Just noting here that some printer firmwares do expect light off delay to
include movement times, and others don't. I suspect UVTools would need a
way to know which printers require what. For example, I'm pretty sure the
Sonic Moni 4k expects the delay to include travel times, while the Anycubic
Photon M3 Max does not.

WestonHawk, 2022-05-04:
@tslater2006 Thanks for the heads up. I assumed that was the case as well, I should have mentioned it. Wouldn't want to screw over other people to fix my problem lol

All I would need is an option to manually turn this calculation off within the suggestion settings. But yeah, this requires more of a database to be seamless for those unaware of this difference.

sn4k3, 2022-05-04:
> When applying the "Wait time before cure: 20/2s" recommendation it is also taking into account the time to lift and retract, which is not necessary on my printer.

That's correct, just trust UVtools, If is setting LightOff with movements time is because it should to, meaning the format/version you using is unable to use raw Wait time instead. Upgrading to CTBv4 will set Wait time instead of light off delay in the format you advise.

> Problem 2

Again, is a file limitation, if UVtools isn't creating the blank layer is because your format lacks a way to use a layer in same position. With your notes, making me thinking you are on CTBv2 which is very old. You should consider upgrade to V4 along with latest firmware for saturn

WestonHawk, 2022-05-04:
> With your notes, making me thinking you are on CTBv2 which is very old. You should consider upgrade to V4 along with latest firmware for saturn

You found the problem! This is my first time using Lychee in nearly a year and my old installation used CTBv2. After updating to the latest version of Lychee it was still set to CTBv2 and I did not notice. I had already updated my Saturn's firmware though. I have Lychee set to CTBv4 now and everything is being applied correctly!

A couple observations

1. The inserted first blank layer (layer 0) has a single grey pixel in the center. I am assuming this was done on purpose because some printers might skip a truly blank layer. This isn't an issue on my printer though so the option to disable this would be appreciated. I know, the layer 0 exposure time is set to `0.1s` so there isn't a chance for anything to cure, but is still bothers me. It is easy to delete though.

2. I see that you are also changing the lift height for the actual first layer (now layer 1, not layer 0) to `0.1mm @ 65mm/min` for LiftHeight and `0.1mm @ 0mm/min` for RetractHeight. The end result is the bed shouldn't move at all between layers 0 and 1. Clever! I am a little concerned that the bed height might become slightly off after attempting to move such a short distance. I'll let you know if I notice any issues.

WestonHawk, 2022-05-04:
Awesome! Thank you for the insight, you have clearly put a lot of thought into this. I was obviously not thinking it all the way through when it came to the movement. It prints 0.05mm layers, no duh it can move 0.1mm!

Masterpiece indeed!