sn4k3/UVtools #474: NullPointerException with MangOS printer Lychee file

bonezinger, 2022-05-13:
OS: Windows 10 64-bit
Slicer: Lychee 3.7.1
Printer: MangOS printer profile
UVtools version: 3.3.2 X64

**Steps to replicate:**

1. Open Lychee, create a new MangOS printer profile (it's at the bottom of the list)
2. Load any 3d file
3. Slice, save file with .cws extension
4. Open the saved file in UVTools with Partial mode

Doing that, the following error occurs:

```
System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.FileFormats.FileFormat.Sanitize() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 5312
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 3343
   at UVtools.WPF.MainWindow.<>c__DisplayClass104_0.<ProcessFile>b__1() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1453
```

The error does NOT occur if you open the file normally. However, it asks for Layer height, Display width and Display height once you load the file, even if you have set up all values properly in Lychee.

I am not entirely sure whether this is a problem with how Lychee saves the .cws file or with UVTools, since files generated with the Nova printer profiles have no such problem. I've reported this on Lychee's Discord as well.

sn4k3, 2022-05-13:
It may due lychee is using other names for keys or not dump the information at all since cws does not require that information which is crucial to UVtools.
Please attach the file here

bonezinger, 2022-05-13:
Here's the file generated.
https://www.mediafire.com/file/th3kapyvh3jall5/MANGOS2.cws/file

sn4k3, 2022-05-13:
There's a bug on sanitize function which not cover this situation on partial mode. Still i can't open your file on normal mode, with a different error... I will have a look. 

Regarding not having layer height and others is because lychee use another key for the name.
CWS is a mess because each variant use different names...

sn4k3, 2022-05-13:
The problem seens on PNG itself, extracting layer from zip gives:
[image]

But UVtools PNG decoder detects as 2400x3840.
Also lychee use 32bit PNGs while UVtools is all 8bit, but this isn't the problem since extra channels are stripped converted to 8bit.

Open as 32bit also gives: 2400, 3840, 3.
Something is really off here

bonezinger, 2022-05-13:
2048x2048 is the default values of the MangOS printer, it shows up as a square, but I suspect Lychee uses that as a "custom" profile for any unlisted printer, as it's the only one that lets you freely change several parameters that are locked for typical printers.

I've changed the parameters of the profile I'm using to use the same screen size as a Nova Whale2 printer, which is 3840x2400, 192mm x 120mm, but it doesn't look like it's being correctly exported by Lychee. - https://www.mediafire.com/file/ugk9rmbxaq18kd3/MangosProfile-Whale2.lyp/file