sn4k3/UVtools #476: [FEATURE REQUEST] Cross-out or otherwise mark tools that cannot be used with current printer/file

Vexs, 2022-05-16:
Pretty much the title. It's a little annoying to go through the "tools" drop-down to discover half of them are unusable- further, when I'm searching for some functionality, it's difficult to remember which tools are usable and which aren't. If these were marked as such ahead of time, it would be very pleasant. 

## Describe the solution you'd like
Tools to be somehow marked as unusable when they cannot be used.
It would also be nice to know what printers can support these tools. 

## Describe alternatives you've considered
One could simply _remove_ the tools from the menu, but that seems kind of unfriendly.

sn4k3, 2022-05-16:
They can be greyed-out, i will look into that

Vexs, 2022-05-18:
Feels like there were a lot more than 6- nbd, this looks great!

sn4k3, 2022-05-21:
Implemented

closed 2022-05-21 by sn4k3