sn4k3/UVtools #477: [BUG] Compute Issues Error

smcdan32, 2022-05-18:
## System
 - OS: Windows 11 21H2
 - Slicer: Lychee 
 - Printer: Photon Zero (.pwo)
 - UVtools version: 3.4.2

UVtools v3.4.2 X64
Operative system: Microsoft Windows 10.0.22000 X64
Processor: AMD Athlon Silver 3050U with Radeon Graphics   
Processor cores: 2
Memory RAM: 4.18 / 5.92 GB
Runtime: win10-x64
Framework: .NET 6.0.5
AvaloniaUI: 0.10.14
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1366 x 768 @ 100% (Primary) (On this)
    WA: 1366 x 720    UA: 1366 x 720

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

## Describe the bug
When trying to detect issues with the sliced file, the error below is received. 

System.AggregateException: One or more errors occurred. (Object reference not set to an instance of an object.)
 ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.Managers.IssueManager.<>c__DisplayClass16_2.<DetectIssues>b__9(Int64 layerIndexInt) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Managers\IssueManager.cs:line 219
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass20_0`1.<ForWorker64>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.TaskReplicator.Run[TState](ReplicatableUserAction`1 action, ParallelOptions options, Boolean stopOnFirstFailure)
   at System.Threading.Tasks.Parallel.ForWorker64[TLocal](Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.ThrowSingleCancellationExceptionOrOtherException(ICollection exceptions, CancellationToken cancelToken, Exception otherException)
   at System.Threading.Tasks.Parallel.ForWorker64[TLocal](Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.For(Int64 fromInclusive, Int64 toExclusive, ParallelOptions parallelOptions, Action`1 body)
   at UVtools.Core.Managers.IssueManager.DetectIssues(IslandDetectionConfiguration islandConfig, OverhangDetectionConfiguration overhangConfig, ResinTrapDetectionConfiguration resinTrapConfig, TouchingBoundDetectionConfiguration touchBoundConfig, PrintHeightDetectionConfiguration printHeightConfig, Boolean emptyLayersConfig, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Managers\IssueManager.cs:line 193
   at UVtools.WPF.MainWindow.<>c__DisplayClass204_0.<ComputeIssues>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.Issues.cs:line 501

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Issues'
2. Click on 'Detect Issues'
3. See error

sn4k3, 2022-05-18:
Looks like malformed file with no layer image.
**Attach the problematic file**

smcdan32, 2022-05-18:
No problem. File attached. The file extension .pw0 is not a support format for file upload, so I changed the file extension to txt.  The error is reproducible with all files after updating to version 3.4.2. 

[wheels.txt](https://github.com/sn4k3/UVtools/files/8713094/wheels.txt)

sn4k3, 2022-05-22:
Is this still a problem with latest release?

smcdan32, 2022-05-27:
Sorry for the late reply. Yes, everything appears to be working with the
latest release. Thanks for your help.

closed 2022-05-27 by sn4k3