sn4k3/UVtools #478: [BUG] Hidden property names still displayed in Edit print parameters

bgyarfas, 2022-05-18:
## System
 - OS: macOS 12.3.1 (Using osx_x64 version of UVTools)
 - Slicer: 
 - Printer: 
 - UVtools version: 3.4.2

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:
UVtools v3.4.2 X64
Operative system: Darwin 21.4.0 Darwin Kernel Version 21.4.0: Fri Mar 18 00:46:32 PDT 2022; root:xnu-8020.101.4~15/RELEASE_ARM64_T6000 X64
Processor: Apple M1 Pro
Processor cores: 10
Memory RAM: 7.07 / 11.66 GB
Runtime: osx.12-x64
Framework: .NET 6.0.3
AvaloniaUI: 0.10.14
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 200% (Primary)
    WA: 2560 x 1415    UA: 1280 x 708
2: 1312 x 848 @ 200%
    WA: 1312 x 818    UA: 656 x 409
3: 1080 x 1920 @ 200%
    WA: 1080 x 1895    UA: 540 x 948

Path:       /private/var/folders/8_/5lqtm8v914g72cbf7mkxfvlw0000gy/T/AppTranslocation/A5EEA1D7-F9A9-4046-B087-4B9E1338E314/d/UVtools_v3.4.2.app/Contents/MacOS/
Executable: /private/var/folders/8_/5lqtm8v914g72cbf7mkxfvlw0000gy/T/AppTranslocation/A5EEA1D7-F9A9-4046-B087-4B9E1338E314/d/UVtools_v3.4.2.app/Contents/MacOS/UVtools

```

## Describe the bug
In the "Edit print parameters" dialog, with the "Change settings per a layer range" selected, the property names of the hidden properties are all displayed in the first row.

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'Edit print parameters'
2. Click on 'Change settings per a layer range'

## Expected behavior

## Screenshots

[image]

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2022-05-18:
All those settings can really exist in some file formats, eg: gcode ones.
The lacking of bottom properties show that present fields are correct, with exception of overlaped "Bottom layer count" which also overlaps to me, i will look into it.

What is your file extension here?

bgyarfas, 2022-05-18:
This was from a ctb v4 exported from lychee 3.7.101. It's worth noting that in UVTools 3.4.1 the overlapping doesn't occur. Maybe something with the bump in AvaloniaUI or .NET version?

sn4k3, 2022-05-21:
Fixed

closed 2022-05-21 by sn4k3