sn4k3/UVtools #479: [BUG] Wait time before cure does not appear to create empty base layer

Vexs, 2022-05-20:
## System
 - OS: Win10 x64
 - Slicer: Lychee
 - Printer: Elegoo saturn 
 - UVtools version: 3.4.2

UVtools v3.4.2 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: AMD Ryzen 7 2700X Eight-Core Processor         
Processor cores: 16
Memory RAM: 14.09 / 15.93 GB
Runtime: win10-x64
Framework: .NET 6.0.5
AvaloniaUI: 0.10.14
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1410    UA: 2560 x 1410

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

## Describe the bug
Applying the "wait time before cure" suggestion does not appear to create the empty base layer required for this to work- both in-editor and on-printer. I also believe it's skipping the rest time adjustment as well as it appears to instantly start exposing on-printer. 

This bug does not appear with the script that predates this being integrated into uvtools

## To Reproduce
Steps to reproduce the behavior:
Apply wait time before cure
Observe that no empty layer is created in-editor

## Expected behavior
An empty layer is created

## Screenshots
Bottom layer, wait time before cure applied
[image]

Vexs, 2022-05-20:
Hm, I'd expect that the [existing script](https://raw.githubusercontent.com/sn4k3/UVtools/master/UVtools.ScriptSample/ScriptDebandingZSample.cs) wouldn't work then- but it does. 
That said- you were correct, it appears lychee is outputting V2 ctb files however! Changing to V4 _does_ produce the desired results, but I don't get why that script would work then unless it's just ignoring the file version and throwing stuff in that could possibly not work....

It might be nice if there was a warning about this in wait time before ready or similar.

sn4k3, 2022-05-20:
> Hm, I'd expect that the [existing script](https://raw.githubusercontent.com/sn4k3/UVtools/master/UVtools.ScriptSample/ScriptDebandingZSample.cs) wouldn't work then- but it does.

Script is outdated and UVtools suggestion is way safer. It's working, but the deal is: CTBv2 don't support custom layer position, meaning layer 0 will print at your layer height, lets say 0.05mm, but layer 1 (the real first layer) will print at 0.10mm instead of 0.05mm because CTBv2 don't support overrides, potential leading to failure if is too tall to adhese to plate. Due to that potential failure point, i don't allow it on those cases

Vexs, 2022-05-20:
Yep, changing to CTB4 format resolved the issue! Closing, but a note or warning if the version is <3 would be a nice add I think.

closed 2022-05-20 by Vexs