sn4k3/UVtools #482: [Small improvement request] Sort issues by area size by default

Lelong-vincent, 2022-05-22:
So this is really a very small improvement, but when looking at issues of the 3d models (island, resin traps, etc), they are sorted by layer by default, and if you want to sort them by area it first sort them in ascending order (smaller area first). I feel that people are more interested by big area issues rather than small one (nobody is going to support all those 1px islands in their slicer).
Could it be possible to sort issues by area by default or at least make it that when sorting by area it first sort them in descending order?
I know this is a picky request sorry hehe.

Otherwise thanks a lot for what you're doing, this tool is such a life changer!

sn4k3, 2022-05-22:
Default sort is: Type -> Layer -> Area (all ASC)
It will depend on each one taste, some like area better, others layer to get the problem in a incremental way to work on the slicer.
Double click on Area column or other will never hurt.
Will not change the default sort, all i can do is place a setting for user to opt the default filter

sn4k3, 2022-06-19:
Implemented

closed 2022-06-19 by sn4k3