sn4k3/UVtools #483: [BUG] Wait time before cure does not remain while saving

Alomphega, 2022-05-22:
## System
 - OS: Windows 11 PRO x64 v21H2
 - Slicer: Chitubox 1.9.2 basic
 - Printer: Saturn
 - UVtools version: 3.4.3

```
UVtools v3.4.3 X64
Operative system: Microsoft Windows 10.0.22000 X64
Processor: Intel(R) Core(TM) i9-10900K CPU @ 3.70GHz
Processor cores: 20
Memory RAM: 7.62 / 63.92 GB
Runtime: win10-x64
Framework: .NET 6.0.5
AvaloniaUI: 0.10.14
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1032    UA: 1920 x 1032
2: 1920 x 1080 @ 100%
    WA: 1920 x 1032    UA: 1920 x 1032

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 20220521+19h00_**************_x8_UVTools.ctb [Version: 4] [Class: ChituboxFile]

```

## Describe the bug
When we re-open a saved UVTools CTB file, the first "wait time before cure" came to "0" instead of the "40" previously and automatically set and saved (and reappears in the "Suggestions" field instead of the "applied suggestions" one).

## To Reproduce
Steps to reproduce the behavior:
1. Go to '[name_of_the_UVTools_saved_CTB_file]' in the Windows explorer
2. Double-click on it and configure Windows to open it within UVTools automatically (a good way)
3. Go to "Suggestions, automations and analyzer" button
5. See that "wait time before cure" advice while it was already done before previously saving the repaired file

## Expected behavior
At the bottom of the screen, it should be shown "Wait time: 40/7/0s|2/1/0s" instead of this ""Wait time: 0/7/0s|2/1/0s",
and of course in applied suggestions we should have already also a checked "Wait time before cure".

## Screenshots
[image]

Hope this will help a bit,
Best Regards, 
Guy

sn4k3, 2022-05-22:
That's normal but still correct and applied.
Reason: CTBv4 don't have a Bottom property for wait times. UVtools give you that feature by set per layer the diferent wait time. When it create the first empty layer, that layer will be 0s wait time. In same session it will know bottom wait time because it was set but if you reload the file it will show as 0, because layer 0 have 0s wait time, and that is what UVtools use to virtual know the bottom wait time.
In the end everything is set the same way as before reload. If you navigate layer by layer you will see on LayerData the wait time is still there.

I can try to fetch the wait time better and not rely only on first wait but on first layer that has pixels

Alomphega, 2022-05-22:
Yes, good idea to rely on first layer that has pixels because it is confusing for now... For instance : what happens now if we reapply the wait time on a file where is was already applied, then save it ?

sn4k3, 2022-05-22:
>  what happens now if we reapply the wait time on a file where is was already applied, then save it 

Nothing, it will just reapply, resulting in same result as before

Alomphega, 2022-05-22:
Ok thanks ! I let this issue open until you process to this update and tell me to close it.

sn4k3, 2022-05-22:
Will auto close once patch.
Thanks

sn4k3, 2022-06-19:
Fixed

closed 2022-06-19 by sn4k3