sn4k3/UVtools #493: [FEATURE REQUEST] Alignment PCB-Exposure

rdu, 2022-06-14:
## Is your feature request related to a problem? Please describe

I use UVtools to expose my PCBs. This is a unique feature that I'm really incredibly happy about because it speeds up my workflow significantly.

However, I have a problem...

When I try to expose the soldermask, the copper traces are not in line with the holes in the soldermask. If both were in the same spot, then I could just align the PCB twice to the corner of the screen and would have a perfect overlay.

Now I have to make it such that I export the two layers each as, manually overlay them with Gimp, and then import them back again.

My guess is that there is no board size specified in the Gerber format, but maybe the board outline layer could be used as a reference?

Perfect would be to be able to select the start coordinates and a repetition with offset (panalize). (I know, this is actually already a separate features request).

sn4k3, 2022-06-14:
The start x,y is always 0, the left,top padding come from pcb placement that should be respected. However i never test for soldermask layer. Zip one sample and send me so i can track down the problem

rdu, 2022-06-14:
After etching, I expose the soldermask ink by moving the board to the exact same spot. I printed myself a template so that I only have to push the board into the top right corner.

I have found my mistake. The program works perfectly. I overlooked that the 2nd time the mirror feature (which I had previously turned off) was on again. I then misinterpreted the resulting image as a misplaced.

Nevertheless, it would be nice if one could repeat the picture. And maybe the mirror button could remember its position. I think I will forget this quite often.... ;-)

sn4k3, 2022-06-14:
> it would be nice if one could repeat the picture.

Tools - Pattern

> And maybe the mirror button could remember its position

Save a profile just with mirror off and set it as default

closed 2022-06-14 by rdu