sn4k3/UVtools #495: [Crash] System.ComponentModel.Win32Exception (0x80004002): Interfaz no compatible

DrEvilMonkey, 2022-06-17:
Program crashes while trying to load the usb test file for anycubic M3 

System.ComponentModel.Win32Exception (0x80004002): Interfaz no compatible
   at Avalonia.Win32.SystemDialogImpl.<>c__DisplayClass2_0.<ShowFileDialogAsync>b__0() in /_/src/Windows/Avalonia.Win32/SystemDialogImpl.cs:line 120
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at UVtools.WPF.MainWindow.OpenFile(Boolean newWindow, FileDecodeType fileDecodeType) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1013
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 278
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 205
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application

# System
UVtools v3.4.3 X64
Operative system: Microsoft Windows 10.0.18363 X64
Processor: AMD Ryzen 5 5600X 6-Core Processor             
Processor cores: 12
Memory RAM: 10.92 / 31.93 GB
Runtime: win10-x64
Framework: .NET 6.0.5
AvaloniaUI: 0.10.14
OpenCV: 4.5.5-237

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

Machine date time: 06/17/2022 00:08:24
    UTC date time: 06/17/2022 05:08:24

sn4k3, 2022-06-19:
Check with latest version, does it still happens?

sn4k3, 2022-07-28:
I've open a issue with Avalonia but they can't reproduce either, can you please participate and help us identifying the problem? https://github.com/AvaloniaUI/Avalonia/issues/8584

sn4k3, 2023-09-17:
Maybe fixed in last version, reopen if needed

closed 2023-09-17 by sn4k3