sn4k3/UVtools #496: [BUG] Incorrectly identified island (bottom of miniature base, parallel to print surface)

deepsol-oba, 2022-06-20:
## System
 - OS: MacOS 10.15.7
 - Slicer: PrusaSlicer 2.4.2 (to create supports and holes), LycheeSlicer 3.7.101 
 - Printer: Anycubic Photon M3 Plus
 - UVtools version: 3.5.0

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that information
[Dumbledore_SL1_up.pwmb.zip](https://github.com/sn4k3/UVtools/files/8936705/Dumbledore_SL1_up.pwmb.zip)
 bellow this line:

UVtools v3.5.0 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Mon Apr 18 21:50:40 PDT 2022; root:xnu-6153.141.62~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz
Processor cores: 8
Memory RAM: 13.54 / 32.58 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.3
AvaloniaUI: 0.10.15
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary)
    WA: 2560 x 1417    UA: 2560 x 1417

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Dumbledore_SL1_up.pwmb [Version: 516] [Class: PhotonWorkshopFile]

## Describe the bug
The bottom of the base is incorrectly identified as a large island.

## To Reproduce
I am printing a miniature with the base upright - the base is parallel to the print platform.
Using PrusaSlicer I generate support and confirm that the supports are touching the base and then export the STL with supports.
[Dumbledore_SL1_up.pwmb.zip](https://github.com/sn4k3/UVtools/files/8936694/Dumbledore_SL1_up.pwmb.zip)

Next step is to import to LycheeSlicer and here I also verify that the supports go all the way to the base.
I slice with LycheeSlicer for my printer without Antialiasing and then load into UVTools.

UV Tools identifies the first layer of the base as a large island.
Scrolling through the layers I can see
Layer 120 - marked as an island
Layer 119 - I can see the supports

## Expected behaviour
Don't identify a layer having supports as an island

## Screenshots
Layer 120 ![image](https://user-images.githubusercontent.com/69177372/174506079-252507ca-e02d-49bd-af7b-c8f990ab615a.png)
Layer 119 ![image](https://user-images.githubusercontent.com/69177372/174506101-2a285a9c-aafc-4ceb-b074-4fd03b64a17a.png)

## Files
Attached
 
## Exceptions Log file
No Crash

## Additional context
N/A

sn4k3, 2022-06-20:
Nothing wrong, it's settings threshold for detection
[image]

deepsol-oba, 2022-06-20:
Thanks, I will give it a go.

closed 2022-06-20 by deepsol-oba