sn4k3/UVtools #498: ability to import STL / export STL files

bman2000, 2022-06-27:
Great program.

a great feature would be the ability to import an STL file so that tools can be used to solidify the model and then export to a slicer. The sliced file can then be imported back into UVtools for checking and any necessary adjustments. 

Thank you

tslater2006, 2022-06-27:
I think processing of STL files directly in UVTools is outside the scope of the program but obviously @sn4k3 can weigh in, we do have the ability to "voxelize" a sliced file back to STL though I don't think its used in any practical use cases, more for the novelty of it.

sn4k3, 2022-06-27:
UVtools have no 3D/mesh capabilities, that require a whole new program and it's not an easy task.
You need to look for alternatives that work with meshes, such as [Autodesk Meshmixer](https://www.meshmixer.com) even then you may find hard to do what you want clean and easy.