sn4k3/UVtools #501: [BUG] Unable to open .cxdlp file exported by chitu box

e0v1a, 2022-07-01:
## System
 - OS: Windows 10 x64 v21H2
 - Slicer: chitu box  v1.9.3
 - Printer: creality halot one plus
 - UVtools version: 3.5.2

```
UVtools v3.5.2 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: 12th Gen Intel(R) Core(TM) i7-12700K
Processor cores: 20
Memory RAM: 14.78 / 31.75 GB
Runtime: win10-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.15
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 7680 x 1440 @ 125% (Primary) (On this)
    WA: 7680 x 1390    UA: 6144 x 1112

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

```

## Describe the bug
I can't use uvtools to open the cxdlp format file exported by chitu box, it prompts that the loading is abnormal and cannot be verified

## Screenshots

[image]

## Files

and this is test file
[test_file.zip](https://github.com/sn4k3/UVtools/files/9025147/test_file.zip)

e0v1a, 2022-07-01:
Yes, every file is (I can provide some more if you need it), and the slice file exported by halot box doesn't have this problem, but as you all know it's neither unstable nor easy to use

[test2.zip](https://github.com/sn4k3/UVtools/files/9030574/test2.zip)

sn4k3, 2022-07-02:
It may be a mistake on chitubox side... Does the file print?

e0v1a, 2022-07-02:
Yes, it can be printed by the machine normally, but cannot be opened with uvtools

sn4k3, 2022-07-02:
Ok, then machine is not validating that field. I can open that validation on the future patch

e0v1a, 2022-07-02:
Ok, thank you very much for making such a great software to improve our printing success rate~

e0v1a, 2022-07-04:
hi @sn4k3 
I also reported this issue to the tech support team of chitu box and they are verifying the issue
A small suggestion if I may:
For slice files in cxdlp format, uvtools is currently the only tool I can find to preview slice files at pixel level, so if the slice file has an exception that will not affect the normal loading of uvtools, can you continue to load it, and then a pop-up warning will pop up after the loading is complete The unusual part? I guess the handling here might be more elegant :)

sn4k3, 2022-07-04:
The checks are in order to prevent malformed files from load. Some errors like this one does not affect anything at it seens, still it don't mean it's correct. 
You will have to wait for the next patch in order to bypass this.

closed 2022-07-05 by e0v1a