sn4k3/UVtools #503: [BUG]

Pesbaskerville, 2022-07-08:
Hi All! Have a wonderfull day)))

UVtools v3.5.3 X64
Operative system: Microsoft Windows 10.0.19043 X64
Processor: Intel(R) Core(TM) i3-7020U CPU @ 2.30GHz
Processor cores: 4
Memory RAM: 5.32 / 7.91 GB
Runtime: win10-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.15
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 125% (Primary) (On this)
    WA: 1920 x 1080    UA: 1536 x 864
2: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]

```

## Describe the bug
I tried to create  layer from Gerber file. And got an problem with lose some elements from original file.
For example, few SMD-components pins doesn't open in UVTools. I tried change some settings and masks in DipTrace and Altium Designer, but both of them working correctly. 
Please, could you check this problem (I attached originals in ZIP)?

##Screenshots

![2022-07-08_161819](https://user-images.githubusercontent.com/108932844/177982535-6fc1792e-3182-46a9-a96b-9375f8aa6607.png)
![2022-07-08_161948](https://user-images.githubusercontent.com/108932844/177982627-c12249fe-faf2-4c0f-bd26-026ae469bcaf.png)

## Files

[Печатка2 с таймером_gerber.zip](https://github.com/sn4k3/UVtools/files/9071566/2._gerber.zip)

Pesbaskerville, 2022-07-11:
Hi!
I deleted all components which are good during preparing of layers. All other components are SMD. And after converting to layer I got traces only, without SMD components. I checked created GERBER in another programs - Altium Designer and Splint Layout. In the file I can see all components and traces like in original SW - DipTrace. Also Ichecked data of creating GERBER file - all my comonents located on the one layer with traces and other (good) components.

[For debug.zip](https://github.com/sn4k3/UVtools/files/9080831/For.debug.zip)
.

Pesbaskerville, 2022-07-12:
Yep, I see all.

sn4k3, 2022-07-14:
Fixed

closed 2022-07-14 by sn4k3