sn4k3/UVtools #504: [Crash] System.OverflowException: Value was either too large or too small for a Decimal.

donkikote, 2022-07-09:
System.OverflowException: Value was either too large or too small for a Decimal.
   at System.Number.ThrowOverflowException(TypeCode type)
   at System.Decimal.DecCalc.VarDecFromR4(Single input, DecCalc& result)
   at System.Decimal.op_Explicit(Single value)
   at UVtools.Core.Suggestions.SuggestionBottomLayerCount.get_IsApplied() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Suggestions\SuggestionBottomLayerCount.cs:line 35
   at UVtools.WPF.MainWindow.PopulateSuggestions(Boolean tryToAutoApply) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.Suggestions.cs:line 64
   at UVtools.WPF.MainWindow.ProcessFile(String fileName, FileDecodeType fileDecodeType, UInt32 actualLayer) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1729
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application

# System
UVtools v3.5.3 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: AMD Ryzen 5 3600X 6-Core Processor             
Processor cores: 12
Memory RAM: 9.00 / 31.95 GB
Runtime: win10-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.15
OpenCV: 4.5.5-237

Path:       E:\D\UVtools_win-x64_v3.5.3\
Executable: E:\D\UVtools_win-x64_v3.5.3\UVtools.exe

Machine date time: 07/09/2022 10:00:54
    UTC date time: 07/09/2022 08:00:54

sn4k3, 2022-07-09:
Can you upload the file which produced this problem?

rojuvi, 2022-07-10:
Sure, this is the one
[UVToolsCrash.zip](https://github.com/sn4k3/UVtools/files/9079708/UVToolsCrash.zip)

sn4k3, 2022-07-11:
I'm unable to open that file, with a error other than yours. Can you check?

rojuvi, 2022-07-11:
It's weird, first time I opened the file I got the firs error. Now when I open it I get this other error:
`System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.FileFormats.ChituboxFile.<>c__DisplayClass215_1.<DecodeInternally>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\ChituboxFile.cs:line 2118
   at UVtools.Core.FileFormats.ChituboxFile.DecodeInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\ChituboxFile.cs:line 2115
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 3351
   at UVtools.WPF.MainWindow.<>c__DisplayClass104_0.<ProcessFile>b__1() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1358`

The same thing happened with this other files. First time it gives the Overflow error, then the following times it throws the NullReferenceException
[UvToolsCrash2.zip](https://github.com/sn4k3/UVtools/files/9087294/UvToolsCrash2.zip)

sn4k3, 2022-07-11:
That may because it is saving the file once open to apply any fix and as file is already malformed it saves a invalid file.
Can you attach a file without open it first (out of slicer) so i can reproduce? Both your A and B trigger same error.

Also where do you sliced the file? Slicer and version? Machine?

sn4k3, 2022-07-11:
If it was a long time ago, very old versions are no longer supported and may cause problems. As file is already altered by UVtools i can't check the version anymore.

closed 2022-08-22 by sn4k3