sn4k3/UVtools #506: [Crash] System.NullReferenceException: Object reference not set to an instance of an object.

matt-child, 2022-07-11:
System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.Skia.SurfaceRenderTarget..ctor(CreateInfo createInfo) in /_/src/Skia/Avalonia.Skia/SurfaceRenderTarget.cs:line 60
   at Avalonia.Skia.DrawingContextImpl.CreateRenderTarget(Size size, Boolean isLayer, Nullable`1 format) in /_/src/Skia/Avalonia.Skia/DrawingContextImpl.cs:line 1025
   at Avalonia.Skia.DrawingContextImpl.CreateLayer(Size size) in /_/src/Skia/Avalonia.Skia/DrawingContextImpl.cs:line 471
   at Avalonia.Rendering.RenderLayers.Update(Scene scene, IDrawingContextImpl context) in /_/src/Avalonia.Visuals/Rendering/RenderLayers.cs:line 19
   at Avalonia.Rendering.DeferredRenderer.UpdateRenderLayersAndConsumeSceneIfNeeded(IDrawingContextImpl& context, Boolean recursiveCall) in /_/src/Avalonia.Visuals/Rendering/DeferredRenderer.cs:line 416
   at Avalonia.Rendering.DeferredRenderer.Render(Boolean forceComposite) in /_/src/Avalonia.Visuals/Rendering/DeferredRenderer.cs:line 359
   at Avalonia.Rendering.DeferredRenderer.Paint(Rect rect) in /_/src/Avalonia.Visuals/Rendering/DeferredRenderer.cs:line 228
   at Avalonia.Controls.TopLevel.HandlePaint(Rect rect) in /_/src/Avalonia.Controls/TopLevel.cs:line 356
   at Avalonia.Win32.WindowImpl.AppWndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/WindowImpl.AppWndProc.cs:line 541
   at Avalonia.Win32.WindowImpl.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/WindowImpl.WndProc.cs:line 33
Category: Non-UI

# System
UVtools v3.5.0 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz
Processor cores: 8
Memory RAM: 15.02 / 15.94 GB
Runtime: win10-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.15
OpenCV: 4.5.5-237

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

Machine date time: 07/11/2022 14:32:09
    UTC date time: 07/11/2022 20:32:09****

matt-child, 2022-07-11:
Here's the Lychee project and the sliced file that UVTools was working on, for reference.
https://drive.google.com/drive/folders/1tudlmkYryK9KH6CxKCaWfdZFg2Y-OpWu?usp=sharing

I was fixing the file and it got about 1/3 of the way through.   I will update fresh and post back

matt-child, 2022-07-11:
Crashed at about 1/3 of the way through again.  I'm going to redo the individual files a bit and change the print settings so the newer revision will be different.
Also perfectly willing to accept that my computer is causing the issue.  it happens from time to time

Emgu.CV.Util.CvException: OpenCV: Failed to allocate 33177600 bytes
   at Emgu.CV.CvInvoke.cveImdecode(IntPtr buf, ImreadModes loadType, IntPtr dst)
   at Emgu.CV.CvInvoke.Imdecode(IInputArray buf, ImreadModes loadType, Mat dst)
   at Emgu.CV.CvInvoke.Imdecode(Byte[] buf, ImreadModes loadType, Mat dst)
   at UVtools.Core.Layers.Layer.get_LayerMat() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Layers\Layer.cs:line 803
   at System.Collections.Concurrent.ConcurrentDictionary`2.GetOrAdd(TKey key, Func`2 valueFactory)
   at UVtools.Core.FileFormats.FileFormat.DrawModifications(IList`1 drawings, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 5630
   at UVtools.Core.Managers.IssueManager.DrillSuctionCupsForIssues(IEnumerable`1 issues, Int32 ventHoleDiameter, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Managers\IssueManager.cs:line 1085
   at UVtools.Core.Operations.OperationRepairLayers.ExecuteInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationRepairLayers.cs:line 466
   at UVtools.Core.Operations.Operation.Execute(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\Operation.cs:line 732
   at UVtools.WPF.MainWindow.<>c__DisplayClass116_0.<RunOperation>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 2127

sn4k3, 2022-07-11:
> Failed to allocate 33177600 bytes

Thats 33gb of ram :O
How many layers are you drilling?

matt-child, 2022-07-11:
it's 2620 layers, which seems normal, but I'm printing on an Epax E10-8K.  both lychee and UV have been significantly slower compared to my Epax X1K - probably because it's roughly 16x the pixels

matt-child, 2022-07-11:
It was also a pretty big fix.  560 errors, lots of resin traps and suction cup drilling as well being done.  I'm probably pushing my system over what it can do with 16gb of RAM

sn4k3, 2022-07-11:
> it's 2620 layers

Thats the total layers. The problem is the drilling. Try to disable the suction cup fix on repair layers, i guess it wont cause the problem anymore

matt-child, 2022-07-12:
ooooh, gotcha.  At this point the project has changed enough that I'm pretty much working with a new dataset.  thanks for the help, gonna close this one out and go from there.

closed 2022-07-12 by matt-child

sn4k3, 2022-07-12:
On the next patch this will be solved and able to process large datasets