sn4k3/UVtools #507: [BUG] can't edit parameters on Jupiter if file was edited by UVTools

ccatlett1984, 2022-07-12:
Repro steps:
1. Slice file in Lychee or Chitubox
(File can be printed on Jupiter and parameters changed during the print.)
2. Edit any parameter in UVTools and save the file.
(File can be printed on Jupiter, but parameters can't be changed)

Also affects other Elegoo printers (Mars 3, Saturn)

sn4k3, 2022-07-12:
You mean with Tool - Edit print parameters?

What happens if you just edit exposure time, save and print?

ccatlett1984, 2022-07-12:
Yes, only thing edited was the normal layer exposure time.

sn4k3, 2022-07-12:
Ok, i don't have either printers to debug this.
Can you send me two files, one original, other with exposure time changed so i can edit the changed one and send you back for you to test with dry runs?

ccatlett1984, 2022-07-12:
Will get them attached to the bug.

ccatlett1984, 2023-02-10:
[Test CTB files.zip](https://github.com/sn4k3/UVtools/files/10711970/Test.CTB.files.zip)

STL, sliced with Chitubox = can edit on printer
STL, sliced with Chitubox, edited with UVTools = can't edit on printer
STL, sliced with Lychee = can't edit on printer.

sn4k3, 2023-02-11:
It's working as expected, CTB disables per layer settings, which will follow global settings (the ones you can change on printer menu).
If flag is set to follow layer table settings you can't change settings on printer menu because each layer have it own settings. This is also true for other formats like anycubic that have such flag

[image]

ccatlett1984, 2023-02-11:
Hmm, I'm not using different per layer settings, can I disable that flag in
UVTools?

sn4k3, 2023-02-11:
> Hmm, I'm not using different per layer settings, can I disable that flag in UVTools?

You have 10 transition layers, that will make 10 layers to fade exposure, for that the flag must be set:

[image]

UVtools auto set that flag before encode, no way to by pass.

But if you set transition layers to 0 on slicer and don't apply UVtools suggestions it will set to 0

ccatlett1984, 2023-02-11:
Ah, that makes sense. Great 👍

ccatlett1984, 2023-02-12:
To confirm, I sliced a file in Chitubox with transition layers, will test tomorrow and report back.

ccatlett1984, 2023-02-13:
Sliced in Chitubox, with Transition layers enabled.
I am able to edit parameters on the printer with this file.

[image]
[TEST.zip](https://github.com/sn4k3/UVtools/files/10726805/TEST.zip)

sn4k3, 2023-02-14:
Yes, but i guess transition layers don't work as expected while printing? They use software transition AFIK

ccatlett1984, 2023-02-21:
Got the chance to do a "dry print" of that file, can confirm that it does actually generate the transition layers.

sn4k3, 2023-02-21:
> Got the chance to do a "dry print" of that file, can confirm that it does actually generate the transition layers.

Can you provide that file?

ccatlett1984, 2023-02-22:
https://github.com/sn4k3/UVtools/files/10726805/TEST.zip

tslater2006, 2023-03-22:
Hey @sn4k3, I've been taking a look at this and it seems from my testing (see the files attached here) that the "PerLayerSettings" field in the CTB does not include the exposure time. In the zip you will find:

* SingleParam-1resin-noTransition_0x00.ctb
  * This is a "Single Parameter Slice" using 1 resin, and no transition layers, has a value of 0x00 for PerLayerSettings
* SingleParam-1resin-yesTransition_0x00.ctb
  * This is a "Single Parameter Slice" using the same resin but with 6 transition layers. has a value of 0x00 for PerLayerSettings
* MultiParam-1resin-yesTransition_0x40.ctb
  * This is a "Multi parameter slice" but only using the same single resin, and 6 transition layers. has a value of 0x40 for PerLayerSettings.
* MultiParam-1resin-noTransition_0x50.ctb
  * This is a "Multi parameter slice" but only using the same single resin, and no transition layers. has a value of 0x50 for PerLayerSettings.

The take aways here are that it seems from Chitubox directly, that if you are using Chitubox basic, or ChituboxPRO (single parameter slicing mode), that "PerLayerSettings" is always 0x00, even if there are transition layers present with different exposure times!

The other interesting note here is that in Multiparameter slice mode with no transition layers, the value is actually 0x50. So i suspect internally this byte is actually a bit field, where 0x40 represents "multiparameter slice" and maybe 0x10 indicates "no transition layers".

So in summary, it looks like UVTool's definition for when "PerLayerSettings" needs to be 0x40 is not aligned with what Chitubox does, specifically it seems CTB allows for exposure settings to differ per layer without that flag set.

Perhaps removing just that parameter from this check (somehow only for CTB files?): https://github.com/sn4k3/UVtools/blob/c3276525584a9230e42a8aefd1977e1d8fa5f62d/UVtools.Core/Layers/Layer.cs#L1021 would be sufficient to get this working more closely

[ChituTests.zip](https://github.com/sn4k3/UVtools/files/11041442/ChituTests.zip)

sn4k3, 2023-03-22:
From what I observed transition layers in CTB works in hardware mode when `PerLayerSettings` is 0 and in software mode when is not 0 (Controlled by layer data) [Note that your **MultiParam-1resin-yesTransition_0x40** has transition on layer exposure time individually, if you hack the file to clean that exposure to global but keep transition layer count field set to positive i guess you won't have transition time at all with PerLayerSettings > 0], this kind of mixture makes like hard when supporting other formats that have that differently.

> So in summary, it looks like UVTool's definition for when "PerLayerSettings" needs to be 0x40 is not aligned with what Chitubox does, specifically it seems CTB allows for exposure settings to differ per layer without that flag set.

Maybe or not, without proper documentation we will never know the real meaning of this stuff. Lychee for example use a fixed value here. From what I observed with Chitubox that also depends on version, the most consistence value I end with is `Version * 0x10` which ends in 0x10, 0x20, 0x30, 0x40, 0x50 depending on version. V5 does not exists yet and we saw 0x50, then you can be right with the bit thing that we can't know for sure.

> Perhaps removing just that parameter from this check (somehow only for CTB files?):

That will solve this issue but create another worse, people who set dynamic exposures CTB will ignore them and use the global.
A better solution is to ignore exposure times only for transition layers.

tslater2006, 2023-03-22:
Both files I shared that use transition layers have those exposures set on the layers individually (as opposed to hardware enforced). And still the PerLayerSettings is 0.

I like your idea though of just not considering the exposure times of transition layers vs any/all layers.

Thanks for the quick reply :)

sn4k3, 2023-03-22:
> Both files I shared that use transition layers have those exposures set on the layers individually (as opposed to hardware enforced). And still the PerLayerSettings is 0.

I know, they keep the values there as safe guard / information, that's correct. But they are never used due PerLayerSettings = 0. If you change them on hex editor and print that won't affect the time for the transition.

> I like your idea though of just not considering the exposure times of transition layers vs any/all layers.

I will try to find an optimal solution for this problem. Anycubic format can also suffer from that but the transition layers are marked as Firmware which UVtools handles differently. 

Also the firmwares are making it badly, instead of a fixed value to tune the settings why not an offset? That would work with everything fixed or dynamic values and solve the problem where using PerLayerSettings can never edit settings...

sn4k3, 2023-03-22:
Fixed

closed 2023-03-22 by sn4k3

ccatlett1984, 2023-03-22:
One thing to note, the only reason Lychee is setting that, is because the Chitu SDK doesn't offer them the control.... And is ALWAYS setting it.