sn4k3/UVtools #513: [FEATURE REQUEST] Add VLR file support for files from Vlare Slicer

nquam, 2022-07-16:
My new Peopoly Forge makes use of a Vlare control board which primarily uses [Vlare slicer](https://www.vlare.net/) and produces VLR files. Looks like there's no support in here for that. Being a new market printer with a pretty new slicer software I've been having issues with my slices and would like to analyze these files. I expect to see this board used more in the future and being one of the only open source slicers it may gain traction pretty quickly.

Vlare says the file format is open and unencrypted so hopefully this is an easy add. The slicer is free so it shouldn't be difficult to create your own test files.

Here's the [Peopoly provided test file](https://drive.google.com/file/d/1zv_aDjtlCzHJ7Z32Dw2de8MK0oF0suEs/view?usp=sharing) for the Forge.

Worth noting this printer does say it has the ability to run Lychee and Voxeldance Tango files but as of yet they have no configuration for the printer.

sn4k3, 2022-08-17:
VLR is the closed format.
OSF is the open format which is now compatible with UVtools v3.6.0+

Check if your printer accept OSF instead

tseven, 2022-09-05:
First off, thanks @sn4k3 for creating UV Tools, I'm new to resin printing and can already see the value in your app.

If UV tools cannot read .vlr files, how can we import a sliced file to import the settings of our printer? Is there another way to get these settings into UV tools?

I do not see a way to export an OSF from Vlare slicer.

tseven, 2022-09-05:
Here is the printer profile Peoply provides for their Forge printer. It has a surprise hidden at the top of the file (not my doing).
[Phenom Forge.zip](https://github.com/sn4k3/UVtools/files/9486904/Phenom.Forge.zip)

nquam, 2022-09-05:
I saw the OSF option and it does work but I'm concerned that maybe somehow the VLR file, should I use it to print, could have issues that go unseen whereas the OSF might not. I haven't printed with an OSF and I don't actually know what file formats my Peopoly Forge will print from but in their software paired with the Forge there were major bugs with the slicer result files and we were instructed to use VLR. I assume an OSF is compatible if it's in Vlare but I'll have to check.

Obviously we could export to OSF, check the print results, then export as VLR if necessary. It's possible for there to be software discrepancies in the slicer but I absolutely need to use this tool until slicers are better at pointing out problems. I wish slicers got better at pointing out possible failures. This tool is extremely useful so whatever I have to do to check my file I'll do.

Not surprised the makers of Vlare haven't gotten back to you. Never heard back from my reports of various bugs and issues. The only way I get feedback is if I go directly to Peopoly and they talk to Vlare.

nquam, 2022-09-05:
> VLR is OSF with exception of a extra unknown field

Well, not surprised but that's really lame since it's still software in beta.

> I wish slicers got better at pointing out possible failures.

> > Me too, in fact UVtools should be an useless tool.

I've only been resin printing for a couple months and this is both the saddest and most useful thing I've learned. That along with automatic supports are terrible. Can't wait for Lychee to add support for Forge. It has to be better. I guess we can close this if you agree. So long as OSF is an export option then that's the solution.

sn4k3, 2022-09-05:
> Well, not surprised but that's really lame since it's still software in beta.

Voxeldance Tango also supports OSF
PrusaSlicer with my printers also supports OSF via UVtools

> Can't wait for Lychee to add support for Forge

Why wait? You can do your custom printer, export and convert with UVtools to OSF. Or export to STL with supports and use anyslicer.

>  I guess we can close this if you agree.

Yes, i will implement vlr if i got the data to

sn4k3, 2022-09-05:
> Pro version is required to alter a lot of the settings including screen size. 

Just find a printer and have same specs as your, use it and export, convert in UVtools.

Otherwise use [PrusaSlicer](https://github.com/sn4k3/UVtools/wiki/Setup-PrusaSlicer)

nquam, 2022-09-27:
The latest version of Vlare actually has a checkbox under slice settings Advanced for "UVTools Conversion" where you can define a place to output a converted file compatible with UVTools in different extension formats. So problem solved?

sn4k3, 2022-09-27:
> The latest version of Vlare actually has a checkbox under slice settings Advanced for "UVTools Conversion" where you can define a place to output a converted file compatible with UVTools in different extension formats. So problem solved?

I think it will always output to the same folder as where you saved your slice (osf).

> Yeah looks like my OSF file would sometimes error when loading in UVTools but I was able to output to CTB format and it loaded in just fine. It took a lot longer to save the conversion but it worked. There are a bunch of file formats though. Is there one that would be "best" for accuracy or speed of loading?

Why are you converting to other file? Isn't your printer compatible with OSF? If so you don't need conversion and you spare that cycle.

nquam, 2022-09-27:
> I think it will always output to the same folder as where you saved your slice (osf).
> You must ask Vlare.

Yeah you point to the UVTools exe for some reason. I got that wrong. Weirdly it's not to automatically open it.

> Why are you converting to other file? Isn't your printer compatible with OSF? If so you don't need conversion and you spare that cycle.

It's not about the printer it's being able to open it in UVTools. Because some of my OSF files would error in UVTools this was necessary to open a fully compatible version. I'm not sure why this is necessary but it was in some slices.

sn4k3, 2022-09-28:
> Weirdly it's not to automatically open it.

Is just to auto convert. But no open.

> Because some of my OSF files would error in UVTools

Thats strange, which error are you getting? Latest version of UVtools?

hotwheels581, 2023-02-01:
I just converted an Anycubic M3 Max to a Vlare control board so have been discussing a few things with Daniel at Vlare/Duobond. I suggested he work with you to bring vlr capability to Uvtools and he sounded very open to it so perhaps contacting them again may be helpful. 

I've found him to be pretty responsive through messaging on Alibaba. Just do a search for the Vlare control board and use the contact manufacturer function.

Currently it has the same flaw the CTBv4 boards have in that it doesn't follow the wait time on the first layer. I've let them know about this but no timeline on if/when it will be updated. I use the "blank" first layer in UVtools on my Mighty 8ks but can't do it on the Vlare/M3Max right now. (I use variable layer heights and when using this function the UVtools export is not available).

sn4k3, 2023-02-02:
You can use OSF with your board, for user it gives same features as VLR.

> but can't do it on the Vlare/M3Max right now.

And you never will if they don't change the file format, currently they don't have per layer data, which means you are locked to global settings. In that aspect vlare is more limited than anycubic format.

hotwheels581, 2023-02-02:
I'd think that it has to have per layer data as it has variable layer height available with separate exposure/lift heights/lift speeds for each layer height option you use on a print or am I missing something? (I am decidedly not a programmer) When using that option it is only savable in the VLR format. It will not save it in OSF or export to UVtools. If you activate variable lift height the export to UVtools option disappears.

I've also used the "extra support exposure" function and one can see it active on the touchscreen with different portions of the of the layer having different exposure times.

Am I missing something with the per layer data comment?

sn4k3, 2023-02-02:
> as it has variable layer height available with separate exposure/lift heights/lift speeds for each layer height option you use on a print 

Where do you saw that? If you can prepare a file out of slicer with that information you can send it to me. Last time I checked VLR format is same as OSF with only 8 more bytes.

> I've also used the "extra support exposure" function and one can see it active on the touchscreen with different portions of the of the layer having different exposure times.

That time is set on the global table, not per layer. Which means you can only a common time per supports and per model.

You can read the spec here: https://github.com/vlare3d/OSF

hotwheels581, 2023-02-02:
vlr test file with .1 and .01 layer heights will be sent once I find how. Should I zip it?

hotwheels581, 2023-02-02:
btw-Variable lift height is accessed in advance functions in the slice settings.
[image]

hotwheels581, 2023-02-02:
https://drive.google.com/file/d/1dd0yELrvGy2Yg7SGBLc9WPhHMM6MLgzt/view?usp=sharing

sn4k3, 2023-02-02:
I donwloaded but I also need to know the settings you used for that variable height. And also do same file but without variable height marked

hotwheels581, 2023-02-02:
Base settings
[image]

Variable height setting screen. You manually enter the heights. Can have more than the two shown in this example.
[image]

Setting for the layers from 0mm to 38.4mm height. 100um
[image]

[image]
Setting for remainder of print at 10um.
[image]

[image]

hotwheels581, 2023-02-02:
Same print sliced at 100um with no variable;
https://drive.google.com/file/d/1APa5v6FgFgqeqXVp7pjpYMHBrCFl0w8l/view?usp=sharing

sn4k3, 2023-02-04:
I can see that they are using a table per height to define those. While i can see that, there are some fields I can't tell what are for. Also image compression is different. If they are interested now (in the past they were not) in having VLR into UVtools please tell them to contact me or document the VLR format like they did with OSF.

sn4k3, 2026-10-08:
Implemented in next patch

closed 2026-10-08 by sn4k3