sn4k3/UVtools #515: [Crash] System.InvalidOperationException: Invalid Arrange rectangle.

tecchese, 2022-07-19:
# Report
```
System.InvalidOperationException: Invalid Arrange rectangle.
   at Avalonia.Layout.Layoutable.Arrange(Rect rect) in /_/src/Avalonia.Layout/Layoutable.cs:line 396
   at Avalonia.Controls.Canvas.ArrangeOverride(Size finalSize) in /_/src/Avalonia.Controls/Canvas.cs:line 202
   at Avalonia.Layout.Layoutable.ArrangeCore(Rect finalRect) in /_/src/Avalonia.Layout/Layoutable.cs:line 676
   at Avalonia.Layout.Layoutable.Arrange(Rect rect) in /_/src/Avalonia.Layout/Layoutable.cs:line 396
   at Avalonia.Layout.LayoutManager.Arrange(ILayoutable control) in /_/src/Avalonia.Layout/LayoutManager.cs:line 308
   at Avalonia.Layout.LayoutManager.ExecuteArrangePass() in /_/src/Avalonia.Layout/LayoutManager.cs:line 277
   at Avalonia.Layout.LayoutManager.InnerLayoutPass() in /_/src/Avalonia.Layout/LayoutManager.cs:line 251
   at Avalonia.Layout.LayoutManager.ExecuteLayoutPass() in /_/src/Avalonia.Layout/LayoutManager.cs:line 171
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application
```

# System
UVtools v3.5.5 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
Processor cores: 8
Memory RAM: 5.54 / 15.89 GB
Runtime: win10-x64
Framework: .NET 6.0.7
AvaloniaUI: 0.10.16
OpenCV: 4.5.5-237

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

Machine date time: 07/19/2022 17:43:16
    UTC date time: 07/19/2022 15:43:16

# Additional information and Workflow
Complete with additional information and the workflow that caused this crash.

This appears always when I try to work with the PCB exposure Tool

sn4k3, 2022-07-19:
This look related to the UI framework.
When does it happen? When enter on the tool? When executing? When closing?
Does it happen with other files as well?

tecchese, 2022-07-20:
To reproduce the issue:

1) add using the PCB exposure tool the attached zipped gbr file
2) check "invert colors" 
3) Load it
4) move it to center
5) minimize the UVTools window (or swap to another running task/program)... et voilà...
[Senza titolo-1.zip](https://github.com/sn4k3/UVtools/files/9146859/Senza.titolo-1.zip)

sn4k3, 2022-07-20:
Does not reproduce to me.
Do you have any window manager utility? Dual screens? It's maximized or normal resizable window?

tecchese, 2022-07-20:
See capture video:

[Video_220720170533.zip](https://github.com/sn4k3/UVtools/files/9151128/Video_220720170533.zip)

sn4k3, 2022-07-20:
Thank you for the video.
I will report this issue with Avalonia, maybe they can have a clue since i can't reproduce nor saw that before.

sn4k3, 2023-09-17:
Maybe fixed in last version, reopen if needed

closed 2023-09-17 by sn4k3