sn4k3/UVtools #518: [BUG] Crashes when trying to open pwmb generated by ChutuBox for Photon Mono X 6K

srozinov, 2022-07-24:
## System
 - OS: Windows 7 x64
 - Slicer: ChituBox Basic v1.9.3
 - Printer: Anycubic Photon Mono X 6K
 - UVtools version: 3.5.5

# Report
```
System.ComponentModel.Win32Exception (0x80070057): Параметр задан неверно.
   at Avalonia.Win32.SystemDialogImpl.<>c__DisplayClass2_0.<ShowFileDialogAsync>b__0() in /_/src/Windows/Avalonia.Win32/SystemDialogImpl.cs:line 120
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at UVtools.WPF.MainWindow.OpenFile(Boolean newWindow, FileDecodeType fileDecodeType) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1015
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application
```

# System
UVtools v3.5.5 X64
Operative system: Microsoft Windows 6.1.7601 Service Pack 1 X64
Processor: Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
Processor cores: 8
Memory RAM: 5.56 / 31.87 GB
Runtime: win7-x64
Framework: .NET 6.0.7
AvaloniaUI: 0.10.16
OpenCV: 4.5.5-237

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

Machine date time: 07/24/2022 18:28:45
    UTC date time: 07/24/2022 10:28:45

## 
Always crashes while opening pwmb file generated by Chitubox. 
Files generated by Lychee for the same printer and same parameters open successfully.

## To Reproduce
Steps to reproduce the behavior:
1. Just open the pwmb file attached. 

## Files

 
## Exceptions Log file

sn4k3, 2022-07-24:
This is not related to UVtools but UI Framework.
Are you loading the file from a USB drive?

srozinov, 2022-07-24:
No I'm loading it from internal SSD... Will try under Windows 10.

srozinov, 2022-07-24:
Oops it worked from USB stick... Surprised! But why doesn't it work from SSD and why it is always read OK from Lychee?..

sn4k3, 2022-07-24:
It's nothing related with slicer or file. It's bugs on UI framework, it's out of my control.

sn4k3, 2022-07-24:
Same bug is listed here: https://github.com/sn4k3/UVtools/issues/495 but that was with USB, it may not be related to a drive at all

srozinov, 2022-07-24:
OK thanks for the solution.

closed 2022-07-24 by srozinov

sn4k3, 2022-07-28:
I've open a issue with Avalonia but they can't reproduce either, can you please participate and help us identifying the problem?: https://github.com/AvaloniaUI/Avalonia/issues/8584

srozinov, 2022-08-03:
> I've open a issue with Avalonia but they can't reproduce either, can you please participate and help us identifying the problem?: [AvaloniaUI/Avalonia#8584](https://github.com/AvaloniaUI/Avalonia/issues/8584)

Yes, of course