sn4k3/UVtools #520: [BUG] Apreture Flashes not imported with PCB Exposure tool

grdrgrd, 2022-07-26:
## System
 - OS: Windows 10 x64 v21H2
 - Slicer: N/A
 - Printer: Anycubic Photon Mono X 6K
 - UVtools version: 3.5.5

```
UVtools v3.5.5 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i5-3470 CPU @ 3.20GHz
Processor cores: 4
Memory RAM: 9.61 / 15.92 GB
Runtime: win10-x64
Framework: .NET 6.0.7
AvaloniaUI: 0.10.16
OpenCV: 4.5.5-237

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       \\homes.mtu.edu\home\Downloads\UVtools_win-x64_v3.5.5\
Executable: \\homes.mtu.edu\home\Downloads\UVtools_win-x64_v3.5.5\UVtools.exe
Loaded file: UVToolsTest2.pwmb [Version: 1] [Class: PhotonWorkshopFile]
```

## Describe the bug
Possibly related to Bug #503. When I import my Gerber file using the PCB Exposure tool, my aperture flashes (Gerber command D03) are not imported.

## To Reproduce
Steps to reproduce the behavior:
1. Open UVTools (GUI)
2. Open a sliced file previously created for my printer (.pwmb in my case)
4. Select Tools -> PCB exposure
5. Add Gerber file (attached), leave all settings default, then generate the traces
6. See the lack of apertures in the imported image

## Expected behavior
In each of the three corners labeled "Fiducial" in the attached Gerber file there should be a 2 mm and a 0.7 mm aperture. These are not displayed in the imported file. 

## Screenshots
This is how I would expect the file to render:
[image]

This is how it actually renders:
[image]

## Files
This is the Gerber file I am trying to import. I had to zip it to upload it, but when I import it I import the unziped file.
[DirectWrite_MasterCoupon_REV3.zip](https://github.com/sn4k3/UVtools/files/9192383/DirectWrite_MasterCoupon_REV3.zip)

sn4k3, 2022-07-26:
Can you prepare a file just with the Fiducial missing dots?

grdrgrd, 2022-07-26:
Here you go:
[DirectWrite_MasterCoupon_REV3_JustFiducials.zip](https://github.com/sn4k3/UVtools/files/9192639/DirectWrite_MasterCoupon_REV3_JustFiducials.zip)

sn4k3, 2022-07-26:
Got the problem, will fix on the next patch

grdrgrd, 2022-07-26:
Sounds good. Thank you!

sn4k3, 2022-07-29:
Fixed

closed 2022-07-29 by sn4k3